mantidproject/mantid #36055: Manual Testing ISIS SANS

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html

warunawickramasingha, 2023-09-13:
When the Reduction selection is changed from 1D to 2D Save Options are changed automatically from CanSAS(1D) to NxCanSAS(1D/2D) as usual. But when RKH(1D/2D) is ticked this behaviour is not seen. Even after unticking RKH(1D/2D) there is no change.

MialLewis, 2023-09-13:
On MacOS, interface hangs for ~30 seconds when editing the first line, or loading a csv batch file.

MohamedAlmaki, 2023-09-13:
In the IDaaS, if you open the ISIS SANS interface and then try to create a random row then copy/paste it, the error dialog will pop up. The error shown in the log is:

Traceback (most recent call last):  File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 532, in _paste_rows_requested    self._call_settings_listeners(lambda listener: listener.on_paste_rows_requested())  File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 274, in _call_settings_listeners    target(listener)  File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 532, in <lambda>    self._call_settings_listeners(lambda listener: listener.on_paste_rows_requested())  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 170, in on_paste_rows_requested    self._presenter.on_paste_rows_requested()  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 798, in on_paste_rows_requested    self._table_model.paste_rows(self._view.get_selected_rows())  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 89, in paste_rows    self._paste_overwrite(row_positions)  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 85, in _paste_overwrite    self.replace_table_entry(pos, copy.deepcopy(entry))  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 117, in replace_table_entry    self.notify_subscribers()  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 218, in notify_subscribers    subscriber.on_update_rows()  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 463, in on_update_rows    self.update_view_from_table_model()  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 483, in update_view_from_table_model    self._view.clear_table()  File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 2087, in clear_table    self.data_processor_table.removeAllRows()Exception: unknown

warunawickramasingha, 2023-09-13:
The instruction under User files section on where to load MaskFile.txt was not clear.

warunawickramasingha, 2023-09-13:
Display mask testing: change in the Phi Limit did not gave any observable change on the  instrument view.

warunawickramasingha, 2023-09-13:
Processing Testing: After clicking Process All, there were 8 files saved at the default save directory as 4 .xml files and 4 .h5 files (No any .nxs files)

MohamedAlmaki, 2023-09-14:
> In the IDaaS, if you open the ISIS SANS interface and then try to create a random row then copy/paste it, the error dialog will pop up. The error shown in the log is:
> 
> Traceback (most recent call last): File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 532, in _paste_rows_requested self._call_settings_listeners(lambda listener: listener.on_paste_rows_requested()) File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 274, in _call_settings_listeners target(listener) File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 532, in self._call_settings_listeners(lambda listener: listener.on_paste_rows_requested()) File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 170, in on_paste_rows_requested self._presenter.on_paste_rows_requested() File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 798, in on_paste_rows_requested self._table_model.paste_rows(self._view.get_selected_rows()) File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 89, in paste_rows self._paste_overwrite(row_positions) File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 85, in _paste_overwrite self.replace_table_entry(pos, copy.deepcopy(entry)) File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 117, in replace_table_entry self.notify_subscribers() File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 218, in notify_subscribers subscriber.on_update_rows() File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 463, in on_update_rows self.update_view_from_table_model() File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 483, in update_view_from_table_model self._view.clear_table() File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 2087, in clear_table self.data_processor_table.removeAllRows()Exception: unknown

I'm not able to reproduce this. Has anyone gotten this error before?

MohamedAlmaki, 2023-09-15:
> On MacOS, interface hangs for ~30 seconds when editing the first line, or loading a csv batch file.
> 
> EDIT: This hang is caused by attempting to connect to the data archive, which time outs.

Do we need to raise an issue for this?

MialLewis, 2023-09-15:
> > On MacOS, interface hangs for ~30 seconds when editing the first line, or loading a csv batch file.
> > EDIT: This hang is caused by attempting to connect to the data archive, which time outs.
> 
> Do we need to raise an issue for this?

No, it occurs because of an issue with how the VPN handles the data archive on my Mac.

MohamedAlmaki, 2023-09-15:
> Display mask testing: change in the Phi Limit did not gave any observable change on the instrument view.

I've retested this and it is working as expected

MohamedAlmaki, 2023-09-15:
All issues have been raised

closed 2023-09-15 by MohamedAlmaki