mantidproject/mantid #36057: Manual Testing Mantid Basics Course

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

* Check the Qt-help docs, built into Workbench, against the online documentation 
* Open up the basic course: http://docs.mantidproject.org/tutorials/mantid_basic_course/index.html#mantid-basic-course
* Check that the information and exercises are up to date and work in the Workbench release candidate

robertapplin, 2023-09-13:
We came across this again https://github.com/mantidproject/mantid/issues/34739

jhaigh0, 2023-09-13:
There should be manual tests for the instrument viewer

jhaigh0, 2023-09-13:
Load ENGINX00228061_log_alarm_data - show sample logs and untick Filtered Data and this crash occurs
https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/07_invalid_sample_logs.html
        
```Stack Trace:
        Traceback (most recent call last):
  File "C:\Users\ekg26817\AppData\Local\mambaforge\envs\68ManualTest\lib\site-packages\mantidqt\widgets\samplelogs\presenter.py", line 127, in filtered_changed
    self.plot_logs()
  File "C:\Users\ekg26817\AppData\Local\mambaforge\envs\68ManualTest\lib\site-packages\mantidqt\widgets\samplelogs\presenter.py", line 110, in plot_logs
    to_plot = [row for row in self.view.get_selected_row_indexes() if self.model.is_log_plottable(self.view.get_row_log_name(row))]
  File "C:\Users\ekg26817\AppData\Local\mambaforge\envs\68ManualTest\lib\site-packages\mantidqt\widgets\samplelogs\view.py", line 243, in get_selected_row_indexes
    return [row.row() for row in self.table.selectionModel().selectedRows()]
AttributeError: 'NoneType' object has no attribute 'selectedRows'```

robertapplin, 2023-09-14:
@jhaigh0 can this be closed now that #36089 is open, and we can't reproduce the other issue?