mantidproject/mantid #36058: Manual Testing M-slice

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/DirectInelastic/MSliceTestGuide.html

SilkeSchomann, 2023-09-12:
We found the following issues:

- A generated script does not reflect a changed legend colour for recoil lines
- Error message when running the following script:
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

# Python Script Generated by Mslice on 2023-09-12 11:17:56
import mslice.cli as mc
import mslice.plotting.pyplot as plt

ws_MAR21335_Ei60meV = mc.Load(Filename='C:\\Users\\eoc57742\\MSLICE550\\MAR21335_Ei60meV.nxs', OutputWorkspace='MAR21335_Ei60meV')

fig = plt.gcf()
fig.clf()
ax = fig.add_subplot(111, projection="mslice")
cut_ws_0 = mc.Cut(ws_MAR21335_Ei60meV, CutAxis="|Q|,0.0,10.0,0.04", IntegrationAxis="DeltaE,-5.0,5.0,0.0", NormToOne=False, IntensityCorrection=False, SampleTemperature=None)
ax.errorbar(cut_ws_0, label="MAR21335_Ei60meV -5.00<E<5.00", color="#1f77b4", marker="o", ls="-", lw=1.5)

from mslice.cli.helperfunctions import hide_a_line_and_errorbars, append_visible_handle_and_label
from mslice.util.compat import legend_set_draggable

# hide lines, errorbars, and legends
handles, labels = ax.get_legend_handles_labels()
visible_handles = []
visible_labels = []

append_visible_handle_and_label(visible_handles, handles, visible_labels, labels, 0)

legend_set_draggable(ax.legend(visible_handles, visible_labels, fontsize='medium'), True)

ax.set_title('', fontsize=12.0)
ax.set_ylabel(r'Signal/#Events', fontsize=10.0)
ax.grid(True, axis='y')
ax.grid(True, axis='x')
ax.set_ylim(bottom=-18.343584245085168, top=131.56881646472806)
ax.set_xlim(left=-0.18399999999999994, right=10.464)
ax.bragg(workspace='MAR21335_Ei60meV_cut(-5.000,5.000)', element='Aluminium')
mc.Show()
```
- Error message when running example script from https://developer.mantidproject.org/Testing/DirectInelastic/MSliceTestGuide.html#the-command-line-interface
- Renaming a workspace in the workspace manager to a name containing a closing bracket (')') causes an error message and the respective workspace cannot be recovered by renaming a second time.

SilkeSchomann, 2023-09-13:
I have opened the following issues:

- https://github.com/mantidproject/mslice/issues/931
- https://github.com/mantidproject/mslice/issues/932
- https://github.com/mantidproject/mslice/issues/933
- https://github.com/mantidproject/mslice/issues/934

closed 2023-09-13 by SilkeSchomann