mantidproject/mantid #36061: Manual Testing Sliceviewer

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html

rbauststfc, 2023-09-12:
Taking x and y cuts and exporting have different axes limits to slice viewer.

rbauststfc, 2023-09-12:
Should add instructions for testing new auto-scaling options.

rbauststfc, 2023-09-12:
If zoom out so that the data is smaller than the plot area, the cursor information widget should go blank when cursor is off the data, the same way it does when the cursor is entirely off the plot.

rbauststfc, 2023-09-12:
3-sigma autoscaling limit is set below the actual minimum of the data

rbauststfc, 2023-09-12:
No data here (linear limits are (0,0)) but showing some dots when colorscale is log

[image]

rbauststfc, 2023-09-12:
MDHistoWorkspace pressing r to export cut of ROI gives error `NoneType object is not subscriptable`

RichardWaiteSTFC, 2023-09-12:
Cutviewer Test

- step 4 - nonorthogonal view is actually enabled and also screenshot below needs to be updated to show this.
- step 16 instructions say `u1 <-> u2` - replace this with `u1 and u2 are swapped`

Specific Test 1

- `md_3D` workspace is from an earlier test but may have been cleared from ADS so direct user to code to create the workspace (or copy it again here)

Specific Test 6

- [110] should be orthogonal to [001] but in 6.8 it isn't (nonorthogonal gridlines appear in test) - have checked and this is a regression from 6.7 possibly related to #35812

RichardWaiteSTFC, 2023-09-13:
Issues created, here's a summary:

Regressions that need to be fixed before release:
- #36080 
- #36087

New bugs (but not regressions)
- #36088
- #36084
 (I thought this was already an issue but I couldn't find it)

Low priority (not worth making an issues - if everyone agrees?):
- Cursor info table not clearing when cursor goes outside extent of data
- Taking x and y cuts and exporting have different axes limits to slice viewer (mentioned in previous testing #35551)

Updates to docs that should go in #36034

Basic Usage

- Add instructions for new colorlimit autoscale options

Cutviewer Test

- step 4 - nonorthogonal view is actually enabled and also screenshot below needs to be updated to show this.
- step 16 instructions say `u1 <-> u2` - replace this with `u1 and u2 are swapped`

Specific Test 1

- `md_3D` workspace is from an earlier test but may have been cleared from ADS so direct user to code to create the workspace (or copy it again here)

rbauststfc, 2023-09-13:
Testing took 1hr 30mins