mantidproject/mantid #36062: Manual Testing Fit Script Generator

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://docs.mantidproject.org/nightly/interfaces/general/Fit%20Script%20Generator.html

jhaigh0, 2023-09-12:
In the first example we have very similar looking plots but our data seems to be scaled by about a factor of 2

jhaigh0, 2023-09-12:
In the second example, has a similar result where the plots looked the same but the axes limits were a bit different

robertapplin, 2023-09-13:
Opened an issue to update these screenshots https://github.com/mantidproject/mantid/issues/36085

closed 2023-09-13 by robertapplin