mantidproject/mantid #36063: Manual Testing ALF

robertapplin, 2023-09-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/DirectInelastic/ALFViewTests.html

MohamedAlmaki, 2023-09-12:
When attempting to open ALFView in Ubuntu, the program may become unresponsive and take some time to load

sf1919, 2023-09-12:
I could not replicate this on a Windows Conda install.

sf1919, 2023-09-12:
It will be interesting to see what effect this PR has - https://github.com/mantidproject/mantid/pull/36033

robertapplin, 2023-09-12:
I think this is a common problem with InstrumentViewer being slow. We are happy to close

closed 2023-09-12 by robertapplin