mantidproject/mantid #36076: Crash from using EISFDiffSphereAlky fit function

jhaigh0, 2023-09-13:
Discovered during #36045

**Describe the bug**
Mantid is crashing when trying to use `EISFDiffSphereAlky`. From where the C++ breaks I think it could be a python 3.10 thing.

This is a regression.

**To Reproduce**
- Open `Interfaces` -> `Indirect` -> `Data Analysis`
- Go to the `F(Q) fit` tab
- Change `Fit Type` to `EISFDiffSphereAlky` and observe crash

**Expected behavior**
Should be able to use it as normal.

**Platform/Version (please complete the following information):**
Tested in Ubuntu and Windows.

jclarkeSTFC, 2023-09-14:
This also happened with `BivariateGaussian` in a fitting window. In both cases it was one of the declared attributes of the algorithm seemingly triggering the problem.

MohamedAlmaki, 2023-09-15:
@robertapplin is this still in progress?

PR #36100 (2023-09-18): Fix bug when trying to select the EISFDiffSphereAlky function
**Description of work**
Fixes an access violation exception when selecting the `EISFDiffSphereAlky` function in the Indirect - Data Analysis window. The same problem could also occur in other ways, e.g. trying to use a `BivariateGaussian` function in a fitting window. It seemed to be any fitting function that has attributes defined could trigger the access violation.

**Purpose of work**

Fixes #36076 

**Summary of work**

- Acquired GIL before calling into the Boost python interface methods.
- Tidied up `EISFDiffSphereAlky.py`slig [... cut]
closed 2023-09-18 by sf1919