mantidproject/mantid #36077: Review workspace groups manual testing

rbauststfc, 2023-09-13:
We should review the value of the Workspace Groups manual testing to confirm what we really need to be manually checking. At the moment the testing instructions largely just involve running the documentation examples at http://docs.mantidproject.org/concepts/WorkspaceGroup.html, however the doctests will now be providing at least some coverage for this.

robertapplin, 2023-09-28:
A good idea to make sure we cover all these tests in the DocTests and then remove this manual testing group

robertapplin, 2023-10-09:
The majority of example code snippets in the attached Workspace Group documentation are doc tests. The few that are not (the last two code snippets), test the deletion of workspaces in a Workspace Group. This is already tested in the unit tests for WorkspaceGroup https://github.com/mantidproject/mantid/blob/305cc1a68cee330d51bf279903613d4799526727/Framework/API/test/WorkspaceGroupTest.h#L247

I would therefore agree that this manual testing issue doesn't add enough value, and should be removed

closed 2023-11-03 by MialLewis