mantidproject/mantid #36078: ISIS energy transfer tab save section not unlocking

jhaigh0, 2023-09-13:
**Original reporter:** Spencer

**Describe the bug**
In the `ISIS Energy Transfer` tab of the Indirect data reduction interface, after running a successful run, the save button and check boxes remain deactivated (greyed out)

This is a regression.

**To Reproduce**

Follow the second test
https://developer.mantidproject.org/Testing/IndirectInelastic/IndirectInelasticAcceptanceTests.html

After running successfully, observe the save section is deactivated.

**Expected behavior**
Should work as it did in 6.7

**Platform/Version (please complete the following information):**
Tested on Linux

SpencerHowells, 2023-09-13:
@jhaigh0 See PR #36067

mantid-builder, 2023-09-19:
Fixed by #36067