mantidproject/mantid #36084: Bugs in exporting cuts from MDHisto workspaces in sliceviewer

RichardWaiteSTFC, 2023-09-13:
**Describe the bug**
Found during manual testing #36060

Two bugs in the same area:

1. When exporting ROI cuts in MDHisto workspace: `'NoneType' object is not subscriptable`
2. When exporting line cut plots for MDHisto workspace: `SliceViewerModel.export_cuts_to_workspace_mdevent() missing 1 required positional argument: 'cut'`

Happens whether original MDEvent workspace exists or not.

**To Reproduce**
(1) Follow instructions for exporting cuts for line plots and rectangular selection/ROI tool - step 3
https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#mdhistoworkspace

which actually involves running the basic usage test steps 3 c-f here
https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#matrixworkspace
Using an MDHisto workspace

**Expected behavior**
Would produce workspaces corresponding to the cuts

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

sf1919, 2024-02-20:
Adding the Reported by User label after discussions on #36852

jhaigh0, 2024-02-22:
I could see where the problems was with issue number 2, but wasn't sure how to recreate.

For the first problem, when the model is set up, if the workspace is an `MDHisto`, `self.export_pixel_cut_to_workspace` is set to `self.export_pixel_cut_to_workspace_md` (the same as if the we type is and MDEvent). [here](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/sliceviewer/models/model.py#L77)

`self.export_pixel_cut_to_workspace_md` takes and uses `bin_params` which are gathered via `data_view.dimensions.get_bin_params()` [here](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/sliceviewer/presenters/presenter.py#L245). `get_bin_params()` returns `none` in the case that the dimensions aren't binned, which is the case for MDHisto (I think?). So there is some undefined behaviour here, I might need @RichardWaiteSTFC to have a look and let me know what should be happening.

The actual error comes from `_dimension_limits()` in the model where it claims `bin_params` is an optional type, but there doesn't seem to be any guarding against it being none. In the previous function `_roi_binmd_parameters()` I can see a `bin_params is not None` check, so it looks like that possibility needs to be considered in `_dimension_limits()` too.

sf1919, 2024-02-22:
@zjmorgan tagging you as you also flagged this issue up so you might want to be part of this discussion.

RichardWaiteSTFC, 2024-02-22:
> I could see where the problems was with issue number 2, but wasn't sure how to recreate.
>
> For the first problem, when the model is set up, if the workspace is an `MDHisto`, [...]

I haven't had a chance to look in detail, but the root of the problem at least for (2) might be that you can't call `BinMD` on an MDHisto - you have to use `IntegrateMDHistoWorkspace` - with different type of binning arguments?

MohamedAlmaki, 2024-04-29:
> I haven't had a chance to look in detail, but the root of the problem at least for (2) might be that you can't call `BinMD` on an MDHisto - you have to use `IntegrateMDHistoWorkspace` - with [...]

As Jonathan said, the `get_bin_params` returns `None` value. It returns `None` because one of the dimensions doesn't support dynamic rebinning (calling multiple BinMD calls on the workspace) which its logic is defined [here](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/sliceviewer/models/workspaceinfo.py#L34). 

In [this line](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/sliceviewer/views/dimensionwidget.py#L132) `get_bins_params` will return None because one of the dimensions will be `Dimension` widget not a `DimensionNonIntegrated` widget as only `DimensionNonIntegrated` defines `get_bins`. The Dimension is defined `DimensionNonIntegrated` if `can_rebin` (the value returned by `can_support_dynamic_rebinning`) is `True` as you see in [this line](https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/sliceviewer/views/dimensionwidget.py#L49)

So the solution, possibly, is either we instruct the users to do something to convert all workspace dimensions into non-integrated dimensions, or we handle the bins parameters for the generic `Dimension`.

MohamedAlmaki, 2024-07-22:
First issue has been resolved in #37271

PR #39071 (2025-05-13): Fix an error when exporting Sliceviewer pixel cuts from MDHisto workspaces
### Description of work
This PR fixes a bug encountered when exporting pixel cuts from MDHisto workspaces. The issue arose because the `export_pixel_cut_to_workspace` is assigned to the MDEvent version in the model. I have added a new function for performing pixel cuts from MDHisto workspaces. Additionally, I have added unit tests for the export pixel cut functions.

Fixes #36084. 

### To test:
1- Make the workspace `md_3D_histo`
```
md_4D = CreateMDWorkspace(Dimensions=4, Extents=[0,2,-1,1,-1.5,1.5,-0.25,0.25], Names= [... cut]
commit 7fc77c83 (2025-05-13): Fix an error when exporting Sliceviewer pixel cuts from MDHisto workspaces (#39071)

### Description of work
This PR fixes a bug encountered when exporting pixel cuts from MDHisto
workspaces. The issue arose because the `export_pixel_cut_to_workspace`
is assigned to the MDEvent version in the model. I have added a new
function for performing pixel cuts from MDHisto workspaces.
Additionally, I have added unit tests for the export pixel cut
functions.

Fixes #36084. 

### To test:
1- Make the workspace `md_3D_histo`
```
md_4D = CreateMDWorkspace(Dimensions=4, Extents=[0,2,-1,1,-1.5,1.5,-0.25,0.2 [... cut]
closed 2025-05-13 by jclarkeSTFC