mantidproject/mantid #36087: New autoscale options set color limits outside range of data

RichardWaiteSTFC, 2023-09-13:
**Describe the bug**
Found during manual testing #36061

**To Reproduce**
(1) Make a workspace
```
md_4D = CreateMDWorkspace(Dimensions=4, Extents=[0,2,-1,1,-1.5,1.5,-0.25,0.25], Names="H,K,L,E", Frames='HKL,HKL,HKL,General Frame',Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(InputWorkspace=md_4D, UniformParams='5e5') # 4D data
FakeMDEventData(InputWorkspace=md_4D, UniformParams='1e5,0.5,1.5,-0.5,0.5,-0.75,0.75,-0.1,0.1') # 4D data

# Add a non-orthogonal UB
expt_info = CreateSampleWorkspace()
md_4D.addExperimentInfo(expt_info)
SetUB(Workspace='md_4D', c=2, gamma=120)

md_3D_histo = BinMD(InputWorkspace='md_4D', AlignedDim0='H,-2,2,100', AlignedDim1='K,-1,1,100', AlignedDim2='L,-1.5,1.5,100')
```
(2)  Open `md_3D_histo` in sliceviewer
(3) Check autoscale under the colorbar
(4) Make a note of the min and max
(5) Select 3-sigma from dropdown-box under autoscale checkbox
(6) Note the colorbar minimum is below 0 (the actual minimum counts)

**Expected behavior**
Would probably clip extents to min and max of data?

**Screenshots**

**Platform/Version (please complete the following information):**
 - Introduced this release (v6.8)

**Additional context**

RichardWaiteSTFC, 2023-09-18:
@zjmorgan - just want to check you're OK with the proposed solution (e.g. clipping the colorbar limits to Min/Max)

mantid-builder, 2023-09-19:
Fixed by #36119

closed 2023-09-19 by mantid-builder