mantidproject/mantid #36089: Crash in the invalid sample logs tutorial

jhaigh0, 2023-09-13:
**Describe the bug**
This came up in #35980

Crash explained below.

This is a regression from 6.7. Interestingly I get this warning when showing the sample logs in 6.7 
[image]
So I reckon the python upgrade may have something to do with it.

**To Reproduce**
 - Load `ENGINX00228061_log_alarm_data` from the training course data (I think it was added recently so you might have to redownload)
 - Right click and open the sample logs
 - Logs fail to load and there's a python error in the console (this probably causes a side effect which causes the crash)
 - Untick the `Filtered Data` check box
 - Observe crash

**Expected behavior**
Should not be a crash

**Platform/Version (please complete the following information):**
Tested on Windows and Linux

robertapplin, 2023-09-14:
I've tried to reproduce this on a Windows Debug build with no luck. 

I went back onto IDAaaS GPU nightly and I could reproduce it with this error:
[image]

jhaigh0, 2023-09-14:
I think that TypeError is the important thing. In python 3.10 they stopped implicitly converting from float to int in function calls.

mantid-builder, 2023-09-15:
Fixed by #36094

PR #36094 (2023-09-15): Fix sample log implicit conversion warning
**Description of work**
This PR fixes an implicit conversion warning seen when opening the sample logs of the `ENGINX00228061_log_alarm_data` training data in v6.7 of Mantid. The move to python 3.10 means that this warning has now become an exception in Mantid v6.8.

**To test:**
Download and install the tarbar from this build package from branch into an IDAaaS workspace:
https://builds.mantidproject.org/job/build_packages_from_branch/593/

Link to IDAaaS dev site:
https://dev.analysis.stfc.ac.uk/

Then follow the testing instructions in the attached  [... cut]
closed 2023-09-15 by mantid-builder