mantidproject/mantid #36105: Plots from script are hidden

thomashampson, 2023-09-14:
**Describe the bug**
When running a script generated by the plot script generator, the plot is hidden. It used to be shown by default.

**To Reproduce**
1. Load a workspace and plot a spectrum
2. copy plot script to clipboard using the toolbar button
3. close plot
4. paste script into script editor
5. run script
6. plot is hidden, you have to go to the plots widget to show it.

**Expected behavior**
plot should appear

**Platform/Version (please complete the following information):**
 - OS: all
 - Mantid Version: 6.8 manual testing candidate (regression since 6.7)

robertapplin, 2023-09-15:
Replicate with this script
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
from mantid.plots.utility import MantidAxType

iris26176_graphite002_red = Load(Filename='iris26176_graphite002_red.nxs')

fig, axes = plt.subplots(edgecolor='#ffffff', num='iris26176_graphite002_red-1', subplot_kw={'projection': 'mantid'})
axes.plot(iris26176_graphite002_red, color='#1f77b4', label='iris26176_graphite002_red: spec 1', wkspIndex=0)

plt.show()
```

robertapplin, 2023-09-19:
fixed by https://github.com/mantidproject/mantid/pull/36117

closed 2023-09-19 by robertapplin