mantidproject/mantid #36115: Auto Switching in save options is not working as expected in ISIS SANS Reduction GUI

warunawickramasingha, 2023-09-15:
**Describe the bug**

**To Reproduce**

**Expected behavior**
Switch between 1D and 2D, it should automatically switch between CanSAS and NXcanSAS

**Platform/Version (please complete the following information):**
 - OS: all
 - Mantid Version: 6.7

**Additional context**

rbauststfc, 2023-12-11:
We should discuss this with SANS to confirm the expected behaviour before starting work on this.

warunawickramasingha, 2023-12-11:
> We should discuss this with SANS to confirm the expected behaviour before starting work on this.

This issue was raised while following the test instructions for https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html#id4

PR #36634 (2024-01-15): Always Return to Default Save Options When Reduction Mode is Switched
### Description of work

#### Summary of work
Removes the function where if the save options were changed, then the GUI did not automatically select the most appropriate saving algorithms when changing between 1D and 2D reduction modes. 

Fixes #36115 

#### Further detail of work
In addition, updates the documentation in line with the scientist's requests. Improving the clarity of what each saving algorithm is designed for.

### To test:
1. Build the `dev-docs-html` target.
2. Check the formatting of the changed manual test [... cut]
closed 2024-01-15 by SilkeSchomann