mantidproject/mantid #36116: Copy/paste operations raise an error in ISIS SANS

MohamedAlmaki, 2023-09-15:
In IDaaS, attempting to copy/paste a randomly created row in the ISIS SANS interface triggers an error dialogue with the corresponding log error message.:

Traceback (most recent call last): File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 532, in _paste_rows_requested self._call_settings_listeners(lambda listener: listener.on_paste_rows_requested()) File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 274, in _call_settings_listeners target(listener) File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 532, in self._call_settings_listeners(lambda listener: listener.on_paste_rows_requested()) File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 170, in on_paste_rows_requested self._presenter.on_paste_rows_requested() File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 798, in on_paste_rows_requested self._table_model.paste_rows(self._view.get_selected_rows()) File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 89, in paste_rows self._paste_overwrite(row_positions) File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 85, in _paste_overwrite self.replace_table_entry(pos, copy.deepcopy(entry)) File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 117, in replace_table_entry self.notify_subscribers() File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/table_model.py", line 218, in notify_subscribers subscriber.on_update_rows() File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 463, in on_update_rows self.update_view_from_table_model() File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/run_tab_presenter.py", line 483, in update_view_from_table_model self._view.clear_table() File "/opt/mantidworkbenchnightly/scripts/Interface/ui/sans_isis/sans_data_processor_gui.py", line 2087, in clear_table self.data_processor_table.removeAllRows()Exception: unknown

The error only happened in IDaaS and we couldn't reproduce it on other platforms.

rbauststfc, 2023-09-18:
Thanks for raising this issue @MohamedAlmaki. I've tried to replicate the error on IDAaaS but I'm afraid I don't seem to be able to. Are you able to provide any specific steps for reproducing the error on IDAaaS?

MohamedAlmaki, 2023-09-18:
I couldn't reproduce the error the second time, but I thought it was worth investigating. I believe the error occurred because I entered invalid data in a row and then tried to copy and paste it into the same row. There was only one row in the table

rbauststfc, 2023-09-18:
Thanks for the extra information @MohamedAlmaki, do you remember what you entered that was invalid?

MohamedAlmaki, 2023-09-18:
I've entered some text in the first two columns which are supposed to be run numbers

rbauststfc, 2023-09-18:
Thanks, I've just given that a try and it all seems to be working fine for me on IDAaaS. I've tried quite a few different things to see if I can break the copy and paste functionality and I can't find any issues. It's great to have a record of this error in case we ever see it again, but it doesn't seem to be reproducible at the moment. Are you happy if I close this issue and then we can re-open it in future if we see the issue again and have more information?

MohamedAlmaki, 2023-09-18:
Yes, that makes sense. Thanks for the investigation