mantidproject/mantid #36123: `PDCalibration` is not fitting peaks when using `StartWorkspaceIndex` and `StopWorkspaceIndex`

rboston628, 2023-09-18:
**Describe the bug**

`PDCalibration` is largely a wrapper of `FitPeaks`, which has the option to specify start/stop workspace indices to only fit peaks within specific spectra but not others.  Recent work in PR #35837 exposed these properties to users of `PDCalibration`.  

For unknown  reasons, when used in this way `PDCalibration` is failing to fit all peaks in the indicated spectrum.

**To Reproduce**

1. Load neutron data converted to d-spacing, and load peaks from a CIF file
2. To run correctly on one spectrum, use the `ExtractSpectra` algorithm with a single workspace ID.
3. Pass the output workspace from `ExtractSpectra` to `PDCalibration` and observe the fit peaks -- it should contain all peaks.
4. Now run `PDCalibration` with the original dataset indicating `StartWorkspaceIndex` and `StopWorkspaceIndex` equal to the index used before.
5. Observe the fit peaks.  For some reason, not all are being fit.

**Expected behavior**

It is expected that all of the peaks at this workspace index should be fit.  There should be no difference in output when using the workspace indices, compared to extracting the spectrum first.

**Screenshots**

The below screenshots pertain to data from run number 58882 on SNAP at ORNL.  The data is grouped using the "Column" grouping scheme for SNAPLite, diffraction focused, and finally put into `PDCalibration`.  The below are for group 2.

### fitted peaks using an extracted spectra
[image]

All peaks in the group are fit.

### fitted peaks using workspace indices
[image]

Only the low d-space peaks are fit.

**Platform/Version (please complete the following information):**
 - OS: Red Hat Enterprise Linux Workstation 7.9 (on ORNL analysis cluster)
 - OS Version: Red Hat Enterprise Linux Workstation 7.9
 - Mantid Version 6.8

**Additional context**

This showing up as part of the diffraction calibration process used in SNAPRed, for reducing data collected on SNAP at ORNL.g

rboston628, 2023-09-18:
If using `PDCalibration` with the `StartWorkspaceIndex` and `StopWorkspaceIndex` properties, and *also* setting `PreviousCalibrationTable` to use the "pixel" calibration table, the result is as follows:
[image]
This is obviously not a very good fit, but it suggests the issue might be related to the calibration table methods.

rboston628, 2023-09-20:
For more context, running the same process that generated the errant fit, but supplying the `"PreviousCalibrationTable"` property generates the below:
[image]
This is better, but still awful.  However, it suggests an issue inside the calibration table methods.

rboston628, 2023-09-27:
The peak windows are input in d-spacing, and internal logic converts them to TOF.  Looking at the TOF peak windows, they are not remotely close.  Shown here, input peak centers in green, windows in dark blue, actual spectrum orange, the "fit" is the lighter blue, which is only matching a few low-TOF peaks.
[image]

mantid-builder, 2023-10-06:
Fixed by #36172

closed 2023-10-06 by mantid-builder