mantidproject/mantid #36127: System test to create algorithm dialogs

robertapplin, 2023-09-19:
**Describe the bug**
We have recently seen the following stacktrace appear in the errorreporter several times:

```
Name: Not provided  Email: Not provided
        Additional text:
        Not provided
        Stack Trace:
        Traceback (most recent call last):
  File "/opt/conda/envs/mantid-singularity/lib/python3.8/site-packages/mantidqt/widgets/algorithmselector/widget.py", line 260, in execute_algorithm
    dialog = manager.createDialogFromName(algorithm.name, algorithm.version, None, False, presets, "", enabled)
Exception: unknown
```

It seems there is an unknown exception when creating some of our algorithm dialogs. It would be useful to have a system test which attempts to create our algorithm dialogs (but does not open them). We used to have a system test that did something similar, but it was refactored because it took a long time to run. It was refactored in this PR https://github.com/mantidproject/mantid/pull/35685

The new system test should **not** open the dialogs. This will hopefully mean the test will run quickly.

**To Reproduce**
Create a system test which will create each of the algorithm dialogs and checks there is no exception. Run this system test until it fails (assuming it is an unreliable exception).

**Additional context**

robertapplin, 2023-09-19:
It is likely the exception is caused by providing a preset value for one of the properties

sf1919, 2024-05-30:
Not stale