mantidproject/mantid #36130: Opening mantid from terminal on IDAaaS causes error

MialLewis, 2023-09-19:
**Describe the bug**

When trying to open mantid from the terminal in IDAaaS, an error occurs:

`GLib-GIO-ERROR ... Settings schema 'org.gtk.Settings.FileChooser' does not contain a key named 'show-type-column'.`

See screenshots below.

**Screenshots**

[image]

**Platform/Version (please complete the following information):**
 - OS: Rocky 8

robertapplin, 2023-09-19:
This is not a regression as it happens for Mantid v6.7, and it also happens on Centos7

thomashampson, 2024-02-21:
@MialLewis did you try it with the conda env activated?

robertapplin, 2024-05-10:
Found a temporary fix on my local IDAaaS dev env:
1. Remove schema file with `sudo rm /usr/share/glib-2.0/schemas/org.gtk.Settings.FileChooser.schemas.xml`
2. Recreate this file with the contents:
```
<?xml version="1.0" encoding="UTF-8"?>
<!--
  Copyright © 2010 Christian Persch

  This library is free software; you can redistribute it and/or modify
  it under the terms of the GNU Lesser General Public License as published by
  the Free Software Foundation; either version 2.1, or (at your option)
  any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public License
  along with this library. If not, see <http://www.gnu.org/licenses/>.
-->
<schemalist>

  <enum id='org.gtk.Settings.FileChooser.LocationMode'>
    <value nick='path-bar' value='0'/>
    <value nick='filename-entry' value='1'/>
  </enum>

  <enum id='org.gtk.Settings.FileChooser.SortColumn'>
    <value nick='name' value='0'/>
    <value nick='size' value='1'/>
    <value nick='type' value='2'/>
    <value nick='modified' value='3'/>
  </enum>

  <enum id='org.gtk.Settings.FileChooser.SortOrder'>
    <value nick='ascending' value='0'/>
    <value nick='descending' value='1'/>
  </enum>

  <enum id='org.gtk.Settings.FileChooser.StartupMode'>
    <value nick='recent' value='0'/>
    <value nick='cwd' value='1'/>
  </enum>

  <enum id='org.gtk.Settings.FileChooser.ClockFormat'>
    <value nick='24h' value='0'/>
    <value nick='12h' value='1'/>
  </enum>

  <enum id='org.gtk.Settings.FileChooser.DateFormat'>
    <value nick='regular' value='0'/>
    <value nick='with-time' value='1'/>
  </enum>

  <enum id='org.gtk.Settings.FileChooser.TypeFormat'>
    <value nick='mime' value='0'/>
    <value nick='description' value='1'/>
    <value nick='category' value='2'/>
  </enum>

  <schema id='org.gtk.Settings.FileChooser' path='/org/gtk/settings/file-chooser/'>
    <key name='last-folder-uri' type='s'>
      <default>""</default>
    </key>
    <key name='location-mode' enum='org.gtk.Settings.FileChooser.LocationMode'>
      <default>'path-bar'</default>
      <summary>Location mode</summary>
      <description>
	Controls whether the file chooser shows just a path bar, or a visible entry
[... 89 lines cut ...]
      </description>
    </key>
    <key name="date-format" enum="org.gtk.Settings.FileChooser.DateFormat">
      <default>'regular'</default>
      <summary>Date format</summary>
      <description>
        The amount of detail to show in the Modified column.
      </description>
    </key>
    <key name="type-format" enum="org.gtk.Settings.FileChooser.TypeFormat">
      <default>'category'</default>
      <summary>Type format</summary>
      <description>
        Different ways to show the 'Type' column information.
        Example outputs for a video mp4 file:
        'mime' -> 'video/mp4'
        'description' -> 'MPEG-4 video'
        'category' -> 'Video'
      </description>
    </key>
  </schema>

</schemalist>
```
3. Re-compile with `glib-compile-schemas /usr/share/glib-2.0/schemas/`

robertapplin, 2024-05-10:
The above schema update also seems to fix #36220 . I have contacted supportanalysis for some help

robertapplin, 2024-05-14:
Its possible to open workbench from the terminal using

```
GSETTINGS_SCHEMA_DIR='/opt/mantidworkbenchnightly/share/glib-2.0/schemas' ./bin/workbench
```

robertapplin, 2024-05-15:
The latest available glib2 package on Rocky8 is 2.56.4, however the latest version on the website says 2.81.0 . I therefore think Mantid requires a schema from >2.56.4 , and so we cannot use the system schema installed on the IDAaaS instances. Instead we need to use the schema shipped with the Mantid install

Given that I think only a Mantid developer might want to open mantid from the terminal on IDAaaS, I don't think this requires any action. Mantid developers should just use the following command to open workbench from the terminal

```
GSETTINGS_SCHEMA_DIR='/opt/mantidworkbenchnightly/share/glib-2.0/schemas' ./bin/workbench
```

Mantid works fine on IDAaaS when opened the normal way using the executable - therefore I will close this issue, and post a tip in the #tips slack channel to notify other developers how to open Mantid from the terminal

robertapplin, 2024-05-15:
https://mantid.slack.com/archives/C02K3DNA6/p1715768642381589

closed 2024-05-15 by robertapplin

robertapplin, 2024-05-15:
Or even better, you can open with
```
/opt/mambaforge/bin/conda run --prefix /opt/mantidworkbenchnightly mantidworkbench
```

If you ever want to know what command an Applications Menu shortcut is running you can find them in /etc/xdg/applications/.
For example: /etc/xdg/applications/mantid_workbenchNightly.desktop:
[Desktop Entry]
Name=Mantid Workbench Nightly
Comment=Neutron Data Reduction Framework
GenericName=MantidWorkbench
Exec=/bin/bash -c "source ~/.bashrc && /opt/mambaforge/bin/conda run --prefix /opt/mantidworkbenchnightly mantidworkbench"
Terminal=false
X-MultipleArgs=false
Type=Application
Categories=Mantid;
Icon=mantid_workbench
StartupNotify=false