mantidproject/mantid #36137: New math functions required for hyperbolic trigonometry

rbauststfc, 2023-09-20:
Required to support the Polarised SANS reduction at ISIS.

Requirements for this still need to be gathered.

dehoni, 2023-11-06:
The time evolution of the transmission and polarisation of a polarised 3He analyser follows analytical expressions that involve hyperbolic functions. At least "arccosh" will be needed to calculate the opacity to measure the optical performance of the 3He cell.
The functions will be applied on a Mantid workspace defined and is can be defined as:
def arctanh(x):
    mtd['plus_ws']=1+x
    mtd['minus_ws']=1-x    
    Logarithm('plus_ws',OutputWorkspace='plus_ws')
    Logarithm('minus_ws',OutputWorkspace='minus_ws')
    Minus('plus_ws','minus_ws',OutputWorkspace='arctanh_ws')
    return 0.5*mtd['arctanh_ws']
    DeleteWorkspace('plus_ws')
    DeleteWorkspace('minus_ws')
    
def tanh(x):
    x=2*x
    Exponential(x,OutputWorkspace='exp')
    mtd['minus_ws']=mtd[exp]-1
    mtd['plus_ws']=mtd[exp]+1
    Divide('minus_ws','plus_ws',OutputWorkspace='tanh_ws')
    DeleteWorkspace('minus_ws')
    DeleteWorkspace('plus_ws')   

def arccosh(x):
    Power(x,Exponent=2,OutputWorkspace='ws')
    mtd['ws']=mtd['ws']-1
    Power('ws',Exponent=0.5,OutputWorkspace='ws')
    mtd['ws']=mtd['ws']+x
    Logarithm('ws',OutputWorkspace='ws')
    return mtd["ws"]

cailafinn, 2024-01-10:
https://docs.mantidproject.org/nightly/algorithms/UserFunction1D-v1.html?highlight=userfunction1d

cailafinn, 2024-04-15:
No longer required, functions provided by UserFunction1D & efficiencies being calculated directly.

closed 2024-04-15 by cailafinn