mantidproject/mantid #36138: Algorithm required for labelling spin states of run periods

rbauststfc, 2023-09-20:
For the Polarised SANS reduction, the periods in the run will need to be examined and labelled with their appropriate spin state.

Requirements for this still need to be gathered. We should also confirm why this is required, as  the `PolarizationCorrectionWildes` algorithm does not need this in order to be useful, but there may be other parts of the workflow that need it before the Wildes algorithm step.

dehoni, 2023-11-06:
The PolarizationCorrectionWildes algorithm works only if the spin states are pass into it in a specific order. Full polarization corrections requires four input workspaces. They should be in the input group in the following order: both flippers off, analyzer flipper on, polarizer flipper on, both flippers on. And so on for the other possible configurations.
This order is not necessarily fixed, or always the same for the experimental setup (depending on the technical implementation). Bookkeeping the right order is extremely error prone by hand, and autoassignment would also help to flag early technical issues (failing equipment, depolarisation of cell, failing flipper...). The flipper state can be easily read out from the Metadata, or if it would be easier can be stored as dedicated variable on IBEX that can be looked up (to be decided).

It includes two stages:
Auto Assign Spin States: This function takes a grouped workspace and tries to identify the order in which the spin states are measure. First checking if all necessary spin states are present as expected for PA, half-polarised SANS etc. 
Assert Correct Order of Spin States: reshuffle workspace so that PolarizationCorrectionWildes algorithm works sensible.

rbauststfc, 2023-11-07:
This issue has some cross-over with issue #35067.