mantidproject/mantid #36143: Algorithm required for extrapolating efficiency wavelength dependence

rbauststfc, 2023-09-20:
A new algorithm is required that can extrapolate the wavelength dependence of the efficiency. This is required for the Polarised SANS reduction at ISIS.

Requirements still need to be gathered.

dehoni, 2023-11-07:
The efficiency of the incoming optical components polarizer and flipper will be known from tabulated values. An algorithm is needed to get these values from calibration measurements (straightforward linear combination and ratios of spin-resolved transmission measurements).
The only particular part is the 3He cell. As part of characterising its performance, calculate the “mean gas length” based on two transmission measurements (depolarisedcell, directbeam). In contrast to opacity=gas_length*wavelength the gas length is an intrinsic quantity independent of wavelength, such that one can extract an averaged quantity over the considered wavelength band to improve statistical significance.

adriazalvarez, 2025-08-04:
Dirk says: 

> There has been a concern that InterpolatingRebin would not perform well over sharp features, like a rising edges or dips, and could introduce artefacts from trailing splines. Honestly if this is the case, then the binning in the calibration was most likely not fine enough and would need to be redone more accurately anyway.

So, we will use InterpolateRebin when applying the polarization corrections if the binning from the stored efficiency workspaces differs from that of the experiment. This, in principle, is a step of the wrapper algorithm and does not require a separate algorithm.