mantidproject/mantid #36145: Algorithm required to calculate the position-dependent efficiency on the SANS detector

rbauststfc, 2023-09-20:
This is required for the Polarised SANS reduction at ISIS.

Requirements still need to be gathered.

dehoni, 2023-11-07:
Similiar the algorithm FlatPlateAbsorptionCorrection, which possibly might be used here. The algorithm calculates an approximation of the attenuation due to absorption and single scattering.

dehoni, 2023-12-05:
Clearly separate step as 2nd order correction.

dehoni, 2025-06-22:
From a physical point this is closely related to the more general issue [Implement wide-angle path corrections for transmission calculations #33543](https://github.com/mantidproject/mantid/issues/33543). #33543 does happen for all measurement with a flat or disk like sample. In both instances, the scattered neutron beam will pass through more material (either flat sample #33543, or cylinder filled with 3He gas for #36145) the higher the scattering angle. For a straight-through beam the neutron will see simply the thickness $t$, while for a beam deflected by an angle $\alpha$ the apparent thickness becomes $t^* = t/\cos \alpha$.

adriazalvarez, 2025-08-04:
We may be able to use _SANSWideAngleCorrection_ algorithm to perform this step. Dirk is going to see the feasability of the algorithm on He transmission data.

closed 2026-08-28 by github-actions[bot]