mantidproject/mantid #36152: Policy for using pip packages in Mantid

robertapplin, 2023-09-21:
**Description**
I'm not aware if we have an explicit policy on what should be done when we want to use a pip package in Mantid (specifically for a conda install). The current general feeling surrounding this is:
1. We do not want to include a pip package as a dependency in our conda recipes. <insert reason>
2. It then makes little sense to push this dependency problem downstream by attempting to install the missing pip packages in our DAaaS workspace configuration repo. It seems like bad practise to install dependencies of a software in a completely detached way like this

**Potential solution**
We should strongly encourage our dependencies to be built into conda packages and uploaded to conda-forge. In the case where we do need to use a pip package, we should make it clear that we do not take responsibility for the automatic install of this dependency, and it should be installed by a user. We would therefore need to ensure the following:
- It should be possible to pip install into the Mantid conda environment on IDAaaS (doesn't seem to be possible atm)
- We should provide documentation on how to pip install into the Mantid conda environment
- We should display warnings in the code which link to this documentation page when someone tries to use a pip package which isn't installed

From the top of my head, Mantid has the following pip dependencies `quasielasticbayes` , `quickBayes`, `gofit`

robertapplin, 2023-09-21:
As part of updating documentation, I will be working on these issues:
- https://github.com/mantidproject/mantid/issues/36149 (targeted at release-next)
- https://github.com/mantidproject/mantid/issues/35835 (targeted at main as its dev docs)

AnthonyLim23, 2023-09-21:
Its not possible to add quickBayes until gofit is on conda https://github.com/conda-forge/staged-recipes/pull/23976 so we could do with a workaround. At this rate it is unlikely for quickBayes to be on conda before release.

MialLewis, 2023-09-21:
Do you know where we are with putting `gofit` on conda-forge - is it something we can do instead of relying on the `gofit` team?

AnthonyLim23, 2023-09-21:
I just know the link that I shared. They are happy to look after it, the slow down is conda forge

jfowkes, 2023-09-25:
@MialLewis the `gofit` conda-forge package has been ready for over a week, we are awaiting a review from the conga-forge team so that it can be merged...

MialLewis, 2023-09-25:
> @MialLewis the `gofit` conda-forge package has been ready for over a week, we are awaiting a review from the conga-forge team so that it can be merged...

I saw that, frustrating! I added something to `conda-forg`e last year and it was reviewed on the same day!

jfowkes, 2023-09-25:
Yeah I think we added the wrong label (`python` instead of `python-c`) and there doesn't seem to be away to change how it's tagged which probably doesn't help!

robertapplin, 2023-12-01:
Closing because this is complete -  we have pip policy documentation

closed 2023-12-01 by robertapplin