mantidproject/mantid #36178: Documentation on how to install Mantid using mamba

robertapplin, 2023-09-27:
**Describe the outcome that is desired.**
We want to provide users with instructions on how to install Mantid using mamba because it has a large speed benefit. We need to decide the best way to provide this information to them. 

**Describe any solutions you are considering**
A potential solution might be to:
1. Write a concepts user documentation page about using mamba to install Mantid.
2. We could then link to this documentation page from the Mantid installation page.

There might be a better solution, so have a think before going for the solution above.

sf1919, 2024-02-21:
also see #36829

Despiix, 2024-09-16:
This issue is resolved already: [ https://www.mantidproject.org/installation/index#installation-conda](url)

closed 2024-09-16 by sf1919