mantidproject/mantid #36179: Exception on IDAaaS ALFView after loading data

robertapplin, 2023-09-27:
**Original reporter:** David Voneshen

**Describe the bug**
There is an error after loading a run number in the ALFView interface in IDAaaS. The same problem does not happen on Windows.

**To Reproduce**
1. Create an IDAaaS ALF powder instance
2. Open Mantid 6.7
3. Go to `Manage User Directories` and set search archive to `all`
4. Open `Direct`->`ALFView` interface
5. Load run 86878 in the sample box
6. There is an exception:
```
File "/opt/mantidworkbench6.7/lib/python3.8/site-packages/matplotlib/backends/backend_qt.py", line 344, in resizeEvent
    self.figure.set_size_inches(winch, hinch, forward=False)
  File "/opt/mantidworkbench6.7/lib/python3.8/site-packages/matplotlib/figure.py", line 2967, in set_size_inches
    raise ValueError(f'figure size must be positive finite not {size}')
ValueError: figure size must be positive finite not [ 1.85714286 -0.03061224 ]
```

**Expected behavior**
There should be no exception. The data should load fine as it does on Windows.

**Platform/Version (please complete the following information):**
 - OS: IDAaaS

robertapplin, 2023-09-27:
This error is caused by the squished color bar widget on the "Render" tab of the InstrumentView

robertapplin, 2023-09-27:
I can replicate the problem on Linux if I make sure the color bar is squished before I start the loading

sf1919, 2023-09-27:
Can this also be replicated on the nightly? I think some colour bar work has gone into v6.8

robertapplin, 2023-09-27:
Yes, this is in nightly as well

mantid-builder, 2023-10-03:
Fixed by #36181

PR #36181 (2023-10-03): Discard the Figure resize event if it has a negative size
**Description of work**
This PR ensures that a QResizeEvent for a C++ ColorbarWidget is discarded if the size is negative or zero. If either of these is true, matplotlib shows an exception which threatens to terminate Mantid.

The bug is present in InstrumentView as well and should be fixed now.

**To test:**
Follow the instructions in the attached issue

Fixes #36179

#### Reviewer ####

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments will be [... cut]
closed 2023-10-03 by mantid-builder