mantidproject/mantid #36183: Unreliable crash when opening InstrumentView

robertapplin, 2023-09-28:
**Original reporter:** David Voneshen

**Describe the bug**
There appears to be an unreliable crash which can occasionally happen when loading data on the ALFView interface

The same crash can happen on ALFView when loading data (because this also uses the InstrumentView).

**To Reproduce**

robertapplin, 2023-09-28:
I can't replicate it on Windows Mantid v6.7
Also can't replicate it on Ubuntu Mantid v6.7, so might be a regression

gemmaguest, 2023-10-02:
As @ConorMFinn said this could be related to the GIL. For reference, here are some other GIL related deadlocks we've seen. Note that deadlocks can sometimes exhibit as a crash instead of a hang (I think this is because some mutex's on the C++ side are clever enough to realise a deadlock situation and they crash out rather than hang).

- #33938
  - deadlock when running two threads that are trying to update the log at the same time, when one of them is updating the log from python code 
  - fixed by #33939
- #33894
  - deadlock while using the AlgorithmManager when called from from different threads from C++ and Python code e.g. in the Reflectometry GUI when processing batches while running live data at the same time.
  - fixed by #33895 
- #33863 
  - deadlock between a python algorithm that calls into code that causes an ADS update without releasing the GIL. The ADS notification waits on a Poco lock to notify its observers. Another thread already has the Poco lock and tries to call into python which deadlocks waiting for the GIL.
  - fixed by #33867 
- #33664
  - deadlock with multiple python algorithms running that are triggering python plot/GUI updates
  - fixed by #33756 and #33863
- #30523
  - hang when closing a plot while ADS is refreshing
  - fixed by #30965

mantid-builder, 2023-10-07:
Fixed by #36224

PR #36224 (2023-10-06): Fix unreliable ALFView segmentation fault
**Description of work**
This PR fixes an unreliable segmentation fault when loading a Sample run number in the ALFView interface. The segmentation fault is caused when the main thread queues up a new algorithm to execute on a background thread, but the previous background thread hasn't finished running and destructs the newly added algorithm by accident. This means that when the next background thread starts up to execute the next algorithm, the `IConfiguredAlgorithm_sptr` has already been deleted (causing a seg fault).

The solution was to ensure we l [... cut]
closed 2023-10-07 by mantid-builder