mantidproject/mantid #36205: Investigate SANS2D Instrument View Cylindrical Y

robertapplin, 2023-10-03:
**Original reporter:** Najet Mahmoudi, Stephen King

**Describe the bug**
The cylindrical Y rendering on Mantid 6.7 and nightly is not what is expected for SANS2D's two detectors.

**To Reproduce**

1. Open Workbench
2. Open `Manage User Directories` and turn on Data archive
3. Load the following file 
```
//isis.cclrc.ac.uk/inst$/NDXSANS2D/Instrument/data/cycle_23_3/SANS2D00081862.nxs
```
4. Right click the workspace and `Show instrument`
5. Change the top option to "Cylindrical Y"
6. Check the result against an earlier version of Mantid

**Screenshots**
At the moment they see:
[image]

It is usual this:
[image]

**Platform/Version (please complete the following information):**
 - OS: Found on Windows, but I assume its the same on all others
 - Mantid Version 6.7 and nightly

robertapplin, 2023-10-03:
The issue exists in v6.7 so is not a regression that needs considering for v6.8

rbauststfc, 2023-10-09:
Having a quick look on IDAaaS, one difference between 6.6 and 6.7 is that a new "Maintain Aspect Ratio" option was added to Render -> Display Settings in the Instrument Viewer. If you uncheck this new option in 6.7 the size of the detectors seem to start matching what is rendered in 6.6.

rbauststfc, 2023-10-09:
I'm not sure why, but when I test this on IDAaaS I see slightly different differences to the screenshots above. I get the following:

6.6:
[image]

Nightly:
[image]

Nightly (with Maintain Aspect Ratio unticked):
[image]

RichardWaiteSTFC, 2023-10-09:
I think the `Maintain Aspect Ratio` option was added for SXD so that the square detector arrays appeared square...It seems it is enabled/True by default, but I think it should be False by default and workbench should remember the setting next time you open the instrument view (which it doesn't - it always reverts to default). Would that work for everyone else?

smk78, 2023-10-09:
Thanks for investigating @robertapplin , @rbauststfc and @RichardWaiteSTFC . We were unaware of the `Maintain Aspect Ratio` option.

The SANS2D detectors are also approximately square so maybe there is a sort of logic to leaving the option True, although we would untick it to use the view for the purpose we intended (eg, creating masks).

As an aside on Mantid on IDAaaS doing it's own thing, is was using it last week and the 3D View was always greyed out!

RichardWaiteSTFC, 2023-10-09:
> As an aside on Mantid on IDAaaS doing it's own thing, is was using it last week and the 3D View was always greyed out!

@smk78 The 3D view on IDAaaS requires OpenGL to be enabled (you can do this in the display settings in instrument view or in the workbench settings (General tab, Instrument section) - maybe it's worth checking is set correctly?

rbauststfc, 2024-06-27:
@smk78 I just stumbled across this issue again - is there anything further that you need from us on this, or could this ticket be closed now?

smk78, 2024-07-01:
Just testing Instrument View on W10/x64 with Mantid 6.10.0 I can sometimes* get the CylindricalY view I expect, so from that perspective I will close this ticket.

*and raise some odd behaviour I am seeing with mantidhelp.

closed 2024-07-01 by smk78