mantidproject/mantid #36248: Linux Fitting Smoke Tests  :muscle: :runner:

robertapplin, 2023-10-10:
* To install the tar.xz package for Linux, run `(sudo) tar -xJf mantid-VA.B.C.tar.xz` in a terminal and it will unzip the package in your current working directory. 
* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n test_mantid_6.6
mamba activate test_mantid_6.6
mamba install -c mantid/label/rc1 mantidworkbench --yes
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

gemmaguest, 2023-10-10:
Getting an unhandled exception if I type something non-valid into the user function e.g.
[image]

```
Traceback (most recent call last):
  File "/home/gemma/Downloads/mantidworkbench/lib/python3.10/site-packages/mantidqt/widgets/fitpropertybrowser/fitpropertybrowser.py", line 417, in fitting_done_slot
    ws = AnalysisDataService.retrieve(name)
KeyError: "'' does not exist."
```

```
Error in execution of algorithm Fit:
Error evaluating function "rubbish" for x=-0.055000: Unexpected end of expression at position 0
```

gemmaguest, 2023-10-10:
Otherwise all good in standalone package on Ubuntu 22.04 :+1:

robertapplin, 2023-10-10:
> ```
> Error in execution of algorithm Fit:
> Error evaluating function "rubbish" for x=-0.055000: Unexpected end of expression at position 0
> ```

Is this related to #36265 @MialLewis ?