mantidproject/mantid #36255: IDAaaS Fitting Smoke Tests  :muscle: :runner:

robertapplin, 2023-10-10:
An IDAaaS session link will be shared with you. It will either be a no-GPU or a GPU IDAaaS instance. All IDAaaS-installed packages are Conda installs.

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

MohamedAlmaki, 2023-10-10:
Not a regression, in the fitting tab, if you fit the function and then move the legend the plot will get small

MohamedAlmaki, 2023-10-10:
In the Fit window, changing the x-axis scale to log and then back to linear does not restore the original plot

MohamedAlmaki, 2023-10-10:
issues raised

closed 2023-10-10 by MohamedAlmaki