mantidproject/mantid #36257: IDAaaS Plotting Smoke Tests  :chart_with_upwards_trend: :bar_chart:

robertapplin, 2023-10-10:
An IDAaaS session link will be shared with you. It will either be a no-GPU or a GPU IDAaaS instance. All IDAaaS-installed packages are Conda installs.

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
 - [ ] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [ ] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Change colour map 
 - [ ] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [ ] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ74041` from the ISIS sample data
 - [ ] 3D surface
 - [ ] 3D wire frame  
 - [ ] Check Toolbar buttons

 ## Sliceviewer
 - [ ] Overly long instructions (don't spend  long!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer!

RichardWaiteSTFC, 2023-10-10:
Plot windows, insturment view, sliceviewer and interfaces (e.g. Engineering Diffraction UI) do not have a maximise or minimize buttons
[image]

If you right-click the window title bar the options are greyed out
[image]

The main window does have it 
[image]

RichardWaiteSTFC, 2023-10-10:
Changing color limit of 3D surface plot using figure options (or double click on colorbar) 
```
Traceback (most recent call last):
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/workbench/plotting/propertiesdialog.py", line 50, in on_ok
    self.canvas.draw()
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/matplotlib/backends/backend_agg.py", line 405, in draw
    self.figure.draw(self.renderer)
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/matplotlib/artist.py", line 74, in draw_wrapper
    result = draw(artist, renderer, *args, **kwargs)
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/matplotlib/artist.py", line 51, in draw_wrapper
    return draw(artist, renderer)
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/matplotlib/figure.py", line 3081, in draw
    self.patch.draw(renderer)
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/matplotlib/artist.py", line 51, in draw_wrapper
    return draw(artist, renderer)
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/matplotlib/patches.py", line 589, in draw
    self._draw_paths_with_artist_properties(
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/matplotlib/patches.py", line 574, in _draw_paths_with_artist_properties
    renderer.draw_path(gc, *draw_path_args)
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/matplotlib/backends/backend_agg.py", line 187, in draw_path
    raise OverflowError(msg) from None
OverflowError: Exceeded cell block limit in Agg, however for the following reasons:

- can not split filled path
- path.should_simplify is False

we can not automatically split up this path to draw.

Please manually simplify your path.

```
[image]

RichardWaiteSTFC, 2023-10-10:
Issue (but not a regression - seen in v6.7) with Peak actions in sliceviewer.
(1) open a ws in sliceviewer
(2) Overlay peak table
(3) Click Add Peak in actions at top
(4) Delete peak table  (this should close peak table in sliceviewer)
(5) Click in 2D plot
```
Traceback (most recent call last):
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/matplotlib/cbook/__init__.py", line 307, in process
    func(*args, **kwargs)
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/mantidqt/widgets/sliceviewer/presenters/presenter.py", line 453, in canvas_clicked
    self._peaks_presenter.add_delete_peak(pos)
  File "/home/l1115324/.conda/envs/mantid6.8_env/lib/python3.10/site-packages/mantidqt/widgets/sliceviewer/peaksviewer/presenter.py", line 358, in add_delete_peak
    presenter_active.add_peak(pos)
AttributeError: 'NoneType' object has no attribute 'add_peak'
```

RichardWaiteSTFC, 2023-10-10:
In instrument view - adding a single-crystal peak in side-by-side view adds a more than one peak at a time

RichardWaiteSTFC, 2023-10-10:
Issues raised

closed 2023-10-10 by RichardWaiteSTFC