mantidproject/mantid #36263: Data Archive option not saved between sessions on IDAaaS

robertapplin, 2023-10-10:
**Describe the bug**
The data archive ON option is not saved between sessions on IDAaaS. This is not a regression because it is present for v6.7 as well as the release candidate.

**To Reproduce**
On an IDAaaS instance:
1. Open Workbench
2. Change the data archive option 
3. Close workbench and open again
4. Check the data archive option. It has been reset

**Expected behavior**
The option should be saved across sessions.

**Platform/Version (please complete the following information):**
 - OS: IDAaaS

jclarkeSTFC, 2023-10-11:
I can't reproduce this, is there a particular type of workspace I need to use? I was using an ENGINX workspace. I changed the setting, closed Workbench, reopened Workbench, setting was what I'd changed it to. Is there something else I need to do? The actual setting is saved to the `Mantid.user.properties` file in `~/.mantid`, maybe it's worth checking what happens to that file.

robertapplin, 2023-10-12:
I think @thomashampson could reproduce this

MohamedAlmaki, 2024-02-02:
I couldn't reproduce it also