mantidproject/mantid #36306: Misscalculated Debye-Waller factor in ComputeIncoherentDos.py

fucaline, 2023-10-23:
**Describe the bug**
I noticed that the Debye-Waller factor in the algorithm ComputeIncoherentDos.py is calculated as:
` DWF = np.exp(-2*(qqgridsq*msd))` with `msd` the averaged mean square displacement provided by the user. 

Given the definition from the in the reference page (https://docs.mantidproject.org/nightly/algorithms/ComputeIncoherentDOS-v1.html#algm-computeincoherentdos), the DWF should be:
` DWF = np.exp(-(qqgridsq*msd))`

I think that the factor 2 should not be here.

jhaigh0, 2023-11-27:
> @jhaigh0 @fucaline
> 
> Yes you're right that the factor of 2 should not be there in the code.
> 
> Sorry for the long delay getting back to you - I had a bit of a detour looking into different notations in use.
> 
> @jhaigh0 when you fix this could you also amend the documentation to say that the `msd` parameter [image] is also called the isotropic atomic displacement parameter `Uiso` [image] in Rietveld refinement programs like [GSAS](https://subversion.xray.aps.anl.gov/trac/EXPGUI/wiki/GSASIntro).
> 
> Could you also say in the docs that there is also another type of atomic displacement parameters used in programs like [FullProf](https://www.ill.eu/sites/fullprof/) called `Biso` which is related by:
> 
> [image]
> 
> The source for this information is page 123 of the [GSAS manual](https://subversion.xray.aps.anl.gov/EXPGUI/gsas/all/GSAS%20Manual.pdf).
> 
> My confusion is because whilst these are the parameters which are actually used by crystallography programs, the International Tables for Crystallography actually defines a different parameter given by:
> 
> ![U_{\mathrm{eqiv}}=\frac{1}{3}\langle\hat{u}^2\rangle](https://camo.githubusercontent.com/8d33a4f80c40f97a0e7878ba5dbc20e8323fe029731b0c0dd0768c8cd1d0ccd1/68747470733a2f2f6c617465782e636f6465636f67732e636f6d2f7376672e696d6167653f555f2537422535436d617468726d253742657169762537442537443d2535436672616325374231253744253742332537442535436c616e676c65253543686174253742752537442535453225354372616e676c65)
> 
> As defined by equation [52] in [this paper](https://www.iucr.org/resources/commissions/crystallographic-nomenclature/adp).

Hi @mducle thanks for the reply, on removing the factor of two `ComputeIncoherentDOSTest::test_nonzero_MDS` fails like this:
```
Traceback (most recent call last):
  File "C:\Users\bya67386\work\mantid\Framework\PythonInterface\test\python\plugins\algorithms\ComputeIncoherentDOSTest.py", line 226, in test_nonzero_MDS
    self.assertAlmostEqual(dos_Ys[i], g[i], delta=g[i] * 1e-12)
AssertionError: 1743788954116.2805 != 7.538620364050139e+24 within 7538620364050.139 delta (7.538620364048395e+24 difference)
```

I think that [this line](https://github.com/mantidproject/mantid/blob/main/Framework/PythonInterface/test/python/plugins/algorithms/ComputeIncoherentDOSTest.py#L49) will have to change to fix this. Could you give me a pointer as to how it should be changed.

mducle, 2023-11-27:
What if you remove the factor of 2 in the `np.exp(2 * msd * qs**2)` in the line (line 49) you linked to? (e.g. so it says `np.exp(msd * qs**2)`

jhaigh0, 2023-11-27:
That's passing now thanks for the help :)

PR #36472 (2023-12-07): Fix miscalculated Debye-Waller factor in ComputeIncoherentDos.py
### Description of work

As pointed out by @fucaline the Debye-Waller factor calculation in ComputeIncoherentDos.py was inconsistent with what was stated in the docs. I asked @mducle for his opinion since he implement the algorithm and he helped me with the fix.

Fixes #36306

### To test:

- Try the example usage from the [doc page](https://docs.mantidproject.org/nightly/algorithms/ComputeIncoherentDOS-v1.html#algm-computeincoherentdos)
- Build the algorithm docs and check what I've added displays correctly and makes sense

---
 [... cut]
closed 2023-12-07 by SilkeSchomann