mantidproject/mantid #36373: Investigate using BayesStretch2 and BayesQuasi2 in Indirect Bayes GUI

robertapplin, 2023-11-07:
**Describe the outcome that is desired.**
We want to update the Indirect Bayes interface to use the new `BayesStretch2` and `BayesQuasi2` algorithms.

Before we do this, we should double check to see that we achieve the same answers as that achieved by the `BayesStretch` and `BayesQuasi` algorithms.

**Describe any solutions you are considering**
Perhaps the first step is to write tests that compare the results of these two algorithms to make sure it is working as expected. Then, we can do the switch in the interface.

SpencerHowells, 2023-11-07:
@robertapplin Anthony & I checked the results and there is a graph in the algorithm instrcutions. There is a discrepancy for SE width at low Q but we believe this is due to a bug in the Fortran version.

robertapplin, 2024-05-20:
Not stale