mantidproject/mantid #36382: Investigate squished widgets in Indirect/Inelastic GUIs

robertapplin, 2023-11-08:
**Describe the bug**
I can see when I open an Indirect/Inelastic GUIs there is often at least one widget which is "squished" and hard to read. If no one else can see this, it might be due to the screen resolution on my laptop. For example on the Data Reduction ISIS Energy Transfer:

[image]

On the diffraction interface:
[image]

On Data manipulations interface:
[image]

We would like to investigate why this is happened and fix it before the release of v6.9.

**To Reproduce**
Open one of the interfaces mentioned above

**Expected behavior**
The widgets should not be squished. The widgets should dynamically resize depending on the resolution of your screen.

**Platform/Version (please complete the following information):**
 - OS: Found on Windows

SpencerHowells, 2023-11-08:
I have that problem on my laptop as well.

adriazalvarez, 2023-11-21:
I don't have that problem in any of my external monitors, or in the laptop screen monitor. 
I did, however, have that problem in the temporary laptop they handed me over on my first week. In the temporary laptop the dpi was much higher than the one I have now. So I guess the issue may be related to that. 
Unfortunately, Is going to be hard for me to inspect this further since I can't reproduce the issue. So I'd ask for other person with that problem to look more into it. 

As a note, adding this line of code just before main qapplication is created may help?

```
 try:
           QApplication.setAttribute(Qt.AA_EnableHighDpiScaling)
  except AttributeError:  # Only for Qt>=5.6, <6.
            pass
  try:
            QApplication.setHighDpiScaleFactorRoundingPolicy(Qt.HighDpiScaleFactorRoundingPolicy.PassThrough)
   except AttributeError:
            pass
```

Also, right now we use ver. 5.15 of qt. Maybe upgrading in the future to higher versions will be useful?

robertapplin, 2023-11-21:
Thanks for taking a look. I can replicate it on my laptop fairly consistently - it has a high dpi, so I can pick this up at some point next sprint

PR #36445 (2023-12-11): Unsquish widgets on the Indirect/Inelastic GUI's
### Description of work
This PR fixes a bug where certain widgets on the Indirect/Inelastic were appearing as being "squished" on screens with high resolutions. It was caused by many of the widgets and QGroupBoxes containing the widgets having a minimum and/or maximum vertical and horizontal size set. This is a bad idea because the size is specified in pixels, and so it is not scalable for screens with larger resolutions. It also doesn't scale with the font size inside the widget.

The solution was to remove this manual setting of min and max si [... cut]