mantidproject/mantid #36387: Mantid utilities for testing in other software

peterfpeterson, 2023-11-09:
Create a package that contains functionality for testing mantid code. How this is packaged is to be determined, but keeping it out of the main mantid repository is preferred.

one. amend_config (contextmanager currently in [drtsans.settings](https://code.ornl.gov/sns-hfir-scse/sans/sans-backend/-/blob/next/drtsans/settings.py?ref_type=heads) ) is used in drtsans tests and source code. There are parallel implementations in [pyrs file_util](https://github.com/neutrons/PyRS/blob/7868f2de5cfb72be643969801e84259732021cda/pyrs/utilities/file_util.py#L80)  and [snapred shims](https://github.com/neutrons/SNAPRed/blob/b2f311978a09ac72e836bc2db7f1a253264b2714/src/snapred/backend/shims.py#L8) . This may be best as a method in mantid.kernel that ships with mantid itself.

two. drtsans.settings unique_workspace_dundername creates a unique temporary workspace name. This should be moved into mantid.api  as an extra method on mtd/AnalysisDataService object

three. A wrapper to CompareWorkspaces that is built for the paradigm of being called as

mantid.testing.assert_equal(wksp1, wksp2)

and generating an assertion error analygous to numpy.testing.assert_equal(ndarray1, ndarray2)

four. Fixtures from [drtsans conftext.py](https://code.ornl.gov/sns-hfir-scse/sans/sans-backend/-/blob/next/tests/conftest.py)  should be moved into the package for easier reuse. Specific fixtures that should be moved are

    clean_workspace - deletes a mantid workspace when a test finishes independent of whether the test passes
    temp_workspace_name - creates a unique temporary workspace name that is very unlikely to clash with any other workspace name. It also registers to remove the workspace when the test function finishes.

Items 1-2 would go in mantid conda package. [Item 3](https://github.com/ccm/resource/itemName/com.ibm.team.workitem.WorkItem/3) would go in a mantid.testing subpackage of mantid. 4th item would go in mantid.framework package.

For reference https://docs.pytest.org/en/7.1.x/how-to/fixtures.html#using-fixtures-from-other-projects

sf1919, 2023-12-04:
@peterfpeterson can you add some labels to this please?

sf1919, 2024-01-22:
Has this work been finished?