mantidproject/mantid #36460: Investigate if both File and Workspace tabs required in Elwin

robertapplin, 2023-11-23:
This is blocked by #33319 . #33319 should be implemented first to ensure we do not lose the functionality of loading multiple files all at once.

**Description**
The Elwin tab has the `File` tab and the `Workspace` tab for loading workspaces. Both tabs can be used to load single or multiple workspaces, however we want to only keep the `Workspace` tab because this is also used in the Data Analysis (Curve fitting) interface. The `File` tab and `Workspace` tab can be used to do the same thing, so we want to remove unneeded code and just have a unified approach for loading multiple files in inelastic GUIs.

**Acceptance Criteria**
- #33319 has been completed
- The `File` tab no longer exists on the Elwin interface

SpencerHowells, 2023-11-23:
@robertapplin This was going to be my next suggestion!

robertapplin, 2024-03-28:
Fixed by  #37048

closed 2024-03-28 by robertapplin