mantidproject/mantid #36482: Make email address mandatory for error reports

jhaigh0, 2023-11-28:
**Is your feature request related to a problem? Please describe.**
Most error reports are not actionable since they don't contain contact information to send follow-up questions to. We also can't contact people when their issue has been fixed.

A large scale error reporter Epic is being planned for next year to address all issues; this is a temporary measure to help us deal with the overwhelming amount of unactionable issues.

**Describe the solution you'd like**
Make an email address mandatory for sending an error report.

**Additional context**
This is okay under GDPR since we require the email address to have a useful response.

peterfpeterson, 2023-11-28:
Can we store the email address in the `Mantid.user.properties` file so they only need to type it once? I'm guessing it is fine with GDPR because it is storing their information on their computer.

jhaigh0, 2023-11-28:
If you click the option to save information for next time Mantid will remember the email &/ username that you typed in already.

The name field is a lot harder to track the person down with because most of the time ISIS scientists will just put their first name, assuming we'll all know who that is when often that isn't the case and we have to ask around.

peterfpeterson, 2023-11-28:
`ConfigService.getUsername()` isn't exposed to python (yet), but would be a reasonable default value for the "name" field. We could be even fancier and figure out how to ask the OS for the user's name. On posix it is the result of `getent passwd <username>`. Adding that as a method in `ConfigService` would not be crazy either.

jhaigh0, 2023-11-28:
Yeah, we can have a look into that. We've found on Windows in python you can do:
```
import win32api

display_name = win32api.GetUserNameEx(3)
```
to get the display name (rather than the username because for us at ISIS that is an ID which can't be related back to a specific user by us).

jhaigh0, 2023-11-28:
Just playing around with it some more and, on our IT system at least, `win32api.GetUserNameEx(8)` gives back the email ties to the account.

PR #36504 (2023-12-06): Mandatory email address for error reports
### Description of work

The decision was made by @sf1919 and others that, as a temporary measure, an email address should be required to submit an error report.
Since it no longer serves a purpose, the no-id share button has been removed.

#### Summary of work

Valid email (validated with regex) is required to enable the share button. If the text is not a valid email, the box outline will be red.

A notice has been added about why we've made this change,
```
Error reports without an email address are not actionable as we cannot ask for further details [... cut]