mantidproject/mantid #36486: Frequent crashes in Nightly build

rbauststfc, 2023-11-29:
We're getting several reports of frequent crashes in the current Nightly. It's unclear at the moment exactly what is causing this, but Richard has experienced it when doing the following:

> Basically I was using it a bit like I would on the beamline for a single crystal diffraction experiment:
> 
> - Loading runs (WISH .raw files)
> - Sometimes converting to d-spacing (ConvertUnits)
> - Opening Instrument view
> - Adding single-crystal peaks to a table
> - Optimising UBs using found peaks (CalculateUMatrix, FindUBUsingLatticePArameters, FindUBUsingFFT)
> - Predicting peaks (PredictPeak)
> - Overlaying predicted peaks and found peaks on the instrument view
> - Zooming in on peaks (on insturment and in TOF in the plot in bottom left of instrument view)
> - Changing TOF limits on instrument view
> - Removing peaks using the rubber tool on instrument view
> - Saving and loading UBs (SaveISawUB, LoadISawUB)
> - Running reduction scripts
> 
> I would say most of the crashes were while the instrument view window was open, but there wasn't any specific action that triggered a crash

I'm opening this issue for us to keep recording when we see crashes and any circumstances around them to see if any patterns start to emerge.

rbauststfc, 2023-11-29:
Error report from a crash when plotting, but without contact information or details of how to reproduce:

> Plot function not working.
> 
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/workbench/plugins/workspacewidget.py", line 172, in _do_plot_spectrum
>     plot_from_names(names, errors, overplot, advanced=advanced)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/mantidqt/plotting/functions.py", line 185, in plot_from_names
>     return plot(
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/mantid/plots/plotfunctions.py", line 89, in inner_func
>     return func(workspaces, *args, **kwargs)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/mantid/plots/plotfunctions.py", line 262, in plot
>     return _update_show_figure(fig)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/mantid/plots/plotfunctions.py", line 291, in _update_show_figure
>     fig.canvas.draw()
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/backends/backend_agg.py", line 400, in draw
>     self.figure.draw(self.renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/artist.py", line 95, in draw_wrapper
>     result = draw(artist, renderer, *args, **kwargs)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/artist.py", line 72, in draw_wrapper
>     return draw(artist, renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/figure.py", line 3175, in draw
>     mimage._draw_list_compositing_images(
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/image.py", line 131, in _draw_list_compositing_images
>     a.draw(renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/artist.py", line 72, in draw_wrapper
>     return draw(artist, renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/axes/_base.py", line 3028, in draw
>     self._update_title_position(renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/axes/_base.py", line 2972, in _update_title_position
>     ax.yaxis.get_tightbbox(renderer)  # update offsetText
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/axis.py", line 1337, in get_tightbbox
>     self._update_label_position(renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/axis.py", line 2614, in _update_label_position
>     bboxes, bboxes2 = self._get_tick_boxes_siblings(renderer=renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/axis.py", line 2149, in _get_tick_boxes_siblings
>     tlb, tlb2 = axis._get_ticklabel_bboxes(ticks_to_draw, renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/axis.py", line 1316, in _get_ticklabel_bboxes
>     return ([tick.label1.get_window_extent(renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/axis.py", line 1316, in <listcomp>
>     return ([tick.label1.get_window_extent(renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/text.py", line 959, in get_window_extent
>     bbox, info, descent = self._get_layout(self._renderer)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/text.py", line 378, in _get_layout
>     _, lp_h, lp_d = _get_text_metrics_with_cache(
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/text.py", line 97, in _get_text_metrics_with_cache
>     return _get_text_metrics_with_cache_impl(
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/text.py", line 105, in _get_text_metrics_with_cache_impl
>     return renderer_ref().get_text_width_height_descent(text, fontprop, ismath)
> 
> 
> 
> 
> 
> [9:48](https://mantid.slack.com/archives/CPXMTE49L/p1701208101775609)
> File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/backends/backend_agg.py", line 233, in get_text_width_height_descent
>     font = self._prepare_font(prop)
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/backends/backend_agg.py", line 267, in _prepare_font
>     font = get_font(_fontManager._find_fonts_by_props(font_prop))
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/font_manager.py", line 1349, in _find_fonts_by_props
>     self.findfont(
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/font_manager.py", line 1260, in findfont
>     ret = self._findfont_cached(
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/font_manager.py", line 1411, in _findfont_cached
>     return self.findfont(default_prop, fontext, directory,
>   File "/opt/anaconda/envs/mantid-dev/lib/python3.10/site-packages/matplotlib/font_manager.py", line 1264, in findfont
>     raise ret.klass(ret.message)
> ValueError: Failed to find font DejaVu Sans:style=normal:variant=normal:weight=normal:stretch=normal:size=10.0, and fallback to the default font was disabled
>         Using: workbench 6.9.20231124.1139 on Red Hat Enterprise Linux

rbauststfc, 2023-11-29:
Opening the instrument view window multiple times on a WISH dataset seems to eventually lead to a crash. I've been doing the following:

1) Load a WISH dataset
1) Open the instrument viewer
1) Open a second instrument viewer window

The crash doesn't always occur on the second instrument viewer window, sometimes I have to add a third, but basically opening multiple windows seems to eventually cause a crash. I've tested in 6.8 and I don't see the same issue there.

jclarkeSTFC, 2023-11-29:
I couldn't reproduce this locally on either Windows or Ubuntu, so maybe it's just an IDAaaS issue.

sf1919, 2023-12-13:
Can we get further testing done on this in IDAaaS?

SilkeSchomann, 2023-12-14:
I can reproduce this on IDAaaS both with the nightly and the released version 6.8. I've used a MAPS>Excitations Powder workspace and a Mari dataset, MAR21335_Ei60meV.nxs.

Initially, it took a few attempts of opening Instrument View until Mantid crashed. After the first crash, Mantid immediately crashes on the first attempt.

rbauststfc, 2023-12-14:
@SilkeSchomann, does switching off OpenGL in the default Mantid settings sort the issue for you too?

rbauststfc, 2024-01-12:
Silke, James and I have agreed to test this one over the next few days to assess if it's still an issue. James hasn't been able to replicate it at all. I've had the following experience so far:

11th Jan morning - Mantid crashed once on trying to open the instrument view for the first time on a loaded SANS2D dataset. After that it wasn't possible to replicate the crash on either the SANS2D dataset or when using `LoadEmptyInstrument` to create a WISH dataset.

12th Jan morning - Mantid crashed once on trying to open the instrument view for the first time on a WISH workspace (created using `LoadEmptyInstrument`). Things then worked OK. I then closed and re-opened Mantid and again experienced the crash when trying to open the instrument view for the first time. After that I tried closing and re-opening Mantid a few times but couldn't get the crash to happen again.

So far the crashing seems less frequent than it was when I was testing back in December, but I'm still seeing it at times. I'll keep updating this comment with my log of any crashes over the next few days.

SilkeSchomann, 2024-01-15:
I've tried to replicate the crashes using WISH data since 12th of January. Neither on the 12th nor on the 15th instrument view crashed.

rbauststfc, 2024-01-16:
I've had crashes each day (see comment above), but today (16th Jan) the crashing was much worse again. Mantid crashed the first time I tried to open instrument view. Then, on re-starting, it crashed the third time of trying to open instrument view. Then again on the 1st time after that and finally on the second time. I stopped testing at this point.

sf1919, 2024-01-16:
From reading through this I think the crashes are probably only on Rocky 8 instances of IDAaaS - is that correct?

rbauststfc, 2024-01-16:
I've been testing mostly on non-Rocky 8 instances.

sf1919, 2024-01-16:
That might be where the issue is - my understanding is that IDAaaS are switching everyone over to Rocky 8 because the previous instance they were using is End of Life and no longer supported. 

Let's keep an eye on this but if it is not happening on Rocky 8 then this might be resolved for us at the IDAaaS side of things.

rbauststfc, 2024-01-16:
OK, well I have seen a crash on Rocky 8 yesterday but I've not tested as much there. Over the next few days I'll focus more of my testing on Rocky 8.

rbauststfc, 2024-01-17:
17th Jan morning - I tested on an IDAaaS Rocky 8 POLREF workspace and experienced crashes trying to open the instrument view the first three times I tried. On the fourth attempt I was able to open the instrument view twice without a crash. I stopped testing at that point.

jclarkeSTFC, 2024-01-17:
It looks like it's something to do with project recovery, disabling project recovery fixes the problem. I could reproduce the problem pretty reliably in an IDAaaS workspace that @rbauststfc shared with me, and after disabling project recovery I couldn't make the problem happen again. I did also generate a stack trace from the problem, it's a segmentation fault but the trace isn't particularly useful.

jclarkeSTFC, 2024-01-17:
I've got it to happen on an Ubuntu VM on my laptop by setting the project recovery save interval timer to 2 seconds, so not just an IDAaaS thing.

closed 2024-01-19 by thomashampson