mantidproject/mantid #36493: Unusual results of `SolidAngle`, probably affecting SANS reduction via `Q1D`

SimonHeybrock, 2023-11-30:
We have run into unexpected results by the `SolidAngle` algorithm. Concretely for at least some instruments with horizontal tubes such as SANS2D and ZOOM. As you can see from the screenshots of the instrument view below, the computed solid angle has 2 maxima above and below the beam center. This is unexpected, since moving a horizontal cylinder up or down (i.e., perpendicular to the cylinder axis) should not make it appear larger.

Note that the figures is using `SolidAngle(Method=GenericShape)`, which internally seems to call the same underlying function (`IDetector::solidAngle()`) as `Q1D` uses when enabling solid-angle correction. Solid-angle correction is enabled by default in `Q1D`.

To reproduce:
1. Load a file or IDF (using `LoadEmptyInstrument).
2. Call `SolidAngle` with default settings (`Method=GenericShape`).
3. Open the instrument view for the result.

SANS2D:
[image]
ZOOM:
[image]
LARMOR (looks ok?)
[image]
LOKI detector test @ LARMOR (does not show two maxima, but an odd shift we do not understand right now)
[image]

smk78, 2023-12-01:
This is what LOQ looks like (the correction on the low-angle detector at 4m, in blue below, is negligible and flat, but more pronounced on the high-angle detector at 0.5m). But these are also individually pixelated detectors (a true multiwire gas detector and scintillator detector, respectively), not tube detectors.
[image]

smk78, 2023-12-01:
Two differences between SANS2D/ZOOM and LARMOR:

1. On SANS2D/ZOOM the tubes are horizontal, on LARMOR they are vertical;
2. How the detectors are 'built' in the respective IDF's are different: SANS2D/ZOOM take one approach, LARMOR takes another.

rmdalgliesh, 2023-12-01:
There is something off in the underlying code related to the horizontal and vertical orientation
If I rotate the SANS2D detector tubes and pixels in the IDF I see something sensible.
[image]

rmdalgliesh, 2023-12-01:
On another note.
The wide angle bank on Larmor doesn't look right either. The green stripes correspond to the centres of the 8 panels in the detector. I would expect a smooth function across the detector as a whole.
[image]

cailafinn, 2023-12-04:
Loading the IDF for SANS2D directly results in the correct output. LARMOR and ZOOM were still broken. Have any changes been made to the SANS2D IDF recently? 

[image]

smk78, 2023-12-04:
Which is indeed my suspicion, @SimonHeybrock ...

sf1919, 2023-12-04:
We have managed to replicate the issue as far back as mantidplot v3.9.2. That is a long time for it to go unnoticed.

AndreiSavici, 2023-12-04:
Note that if one does the reduction properly, and use efficiency calibration with an incoherent sample, the solid angle algorithm has no effect `(Data/SA)/(Incoherent/SA)=Data/Incoherent`

peterfpeterson, 2023-12-04:
I have a vague memory of there being an issue of solid angle not accounting of the occlusion of a pixel by its neighbors. I do not recall if this is properly accounted for in the various detector types.

sf1919, 2023-12-13:
It looks like this is working as it should then @smk78 ? If thats the case perhaps this issue could be closed?

rmdalgliesh, 2023-12-13:
> It looks like this is working as it should then @smk78 ? If thats the case perhaps this issue could be closed? What do you think @SimonHeybrock ?

I don't think so @sf1919 unless someone has implemented one of the changes suggested above. It reads as if a change needs to be made somewhere to correct the problem going forward.

jclarkeSTFC, 2023-12-21:
I've got a PR that fixes this problem (#36581) by defaulting to 11 slices, but also adds the option to specify how many to use when calling `SolidAngle`. It's not ready for review yet because I need to fix a bunch of tests where expected outputs have changed, but the actual implementation is finished. If any of you would like to try the fix and need a package build to do so then let me know.

rmdalgliesh, 2024-01-03:
Thanks @jclarkeSTFC I suspect that we should discuss with @cailafinn what we should be checking for the SANS code. I'm not sure we know enough about the underlying calls in the Q1D calls to make an informed decision on what to test. @cailafinn do you  have any thoughts?

cailafinn, 2024-01-05:
> I'm not sure we know enough about the underlying calls in the Q1D calls to make an informed decision on what to test. 

@rmdalgliesh I still need to do a bit more looking into it as the underlying algorithms predate my time working on mantid. Doing a bit of digging in the code documentation seems to suggest that it's mainly used for wide angle transmission corrections, but it may be used elsewhere.

rmdalgliesh, 2024-01-15:
@cailafinn OK, thanks for checking, we'll need to do some more digging. I think I understand why it might be used there but I would have thought it might come up elsewhere. It may be something we should wrap into the work on the polarisation EPIC. There are few other things we should check as part of that project too.

rmdalgliesh, 2024-01-15:
Hi @jclarkeSTFC this sounds like a sensible plan from the SANS point of view for the time being. We should determine how best to get the testing sorted as soon as possible in the next development cycle.

sf1919, 2024-01-22:
I'm closing this as it's been fixed by #36581

SimonHeybrock, 2024-01-22:
Note that (as far as I can tell at a glance), `Q1D` has not been fixed there (but maybe you are tracking this elsewhere).

dehoni, 2025-10-20:
This feature will easily go unnoticed. The instrument calibration will force that the standard follows the expected 1D scattering curve. You can still show that sectors in different directions will not quite line up (but there is always the argument with statistics etc, detector angle coverage). The issue becomes apparrent in a round robin exercise when comparing a second sample that has been measured on two different instrument.