mantidproject/mantid #36559: Blank lines in ISIS SANS batch file cause confusing error on load

rbauststfc, 2023-12-14:
**Original reporter:** this was discovered from a support query from an external user.

**Describe the bug**

If you try to load a batch file with a blank line at the end into the ISIS SANS GUI it will fail with a confusing error message about there being something wrong in the last populated row of the file. This doesn't help the user to resolve the issue themselves. When `.csv` batch files are opened in Excel rather than in Notepad (which I think is the more typical use case) it's not possible to see that there's a blank row.

**To Reproduce**

1) Download this batch file to a convenient location - it contains a blank row at the end: 
[batch_mode_reduction_blank_rows.csv](https://github.com/mantidproject/mantid/files/13671826/batch_mode_reduction_blank_rows.csv)
1) Go to `Interfaces -> SANS -> ISIS SANS`
1) Click Load Batch File and load the file you downloaded in the first step. This should produce the following error: "Loading of the batch file failed. The sample_scatter entry in row 3 seems to be missing."

**Expected behavior**

The error message should explain that the issue is because of blank rows at the end of the file and that these can be removed by opening the file in a text editor. Alternatively, and probably better, blank rows at the end of the file should maybe just be ignored?

I haven't tested what happens if there are blank rows between populated rows in the batch file, so we should test this as well and make sure we're getting a sensible result from the interface.

smk78, 2023-12-14:
The description of the bug is not quite correct.

The supplied .csv batch file does _not_ have a blank row at the end, rather it contains a row of commas. As commas are the field delimiter, when Mantid tries to import the file it is rightly perplexed to find empty fields and is reporting as much. Thus, I consider the error message to be meaningful and accurate.

If you actually try and load a batch file with a blank line, such as this one
[batch_mode_reduction_blank_rows - Copy.csv](https://github.com/mantidproject/mantid/files/13672530/batch_mode_reduction_blank_rows.-.Copy.csv)
everything works as expected. Indeed Mantid even strips out the blank line (ie, it doesn't create a blank row in the table):
[image]

So the more interesting question here is _how_ the first batch file (with the commas) was created in the first place. As a test I just deliberately added a blank row to the table above:
[image]
and then exported the table:
[image]
The file written:
[batch_mode_reduction_blank_rows - Copy2.csv](https://github.com/mantidproject/mantid/files/13672579/batch_mode_reduction_blank_rows.-.Copy2.csv)
does not contain a row of commas! So the bug reported does not seem to be Mantid creating problems for itself.

I note that the Users batch file has a comment header. This is not something that Mantid writes either. Thus, I think what we are looking at here is a case of a User trying to be too clever for their own good; the problem they are reporting was one of their own making. But, as I say above, the error that Mantid threw was meaningful.

Unless anyone has other thoughts, I am inclined to suggest we close this comment as Won't Fix?

rbauststfc, 2023-12-14:
Happy to close this if not useful/worth fixing. I think the main point was that the user was confused by the error message because it suggests the issue is in a row of the batch file that is actually OK. To solve the problem it was necessary to delete the "empty" rows (not technically empty when viewed in a text editor, but would appear empty when viewed in Excel - not sure how this state was created), which perhaps the user would have been able to do themselves if the error message had been clearer. But if this is very edge case then happy to close if not worth the effort to investigate and/or fix.

rbauststfc, 2023-12-14:
I think the example file I've mocked up and attached to this issue perhaps doesn't demonstrate this very well because the first line in this particular file is a comment, so technically the issue is with the third line of data (although the fourth line of the file). However, if you remove the comment so that there are only two lines in the file plus the final line of commas then the error message changes to say the issue is in the second line of the file, which perhaps makes it clearer where the confusion creeps in.

smk78, 2023-12-14:
Ok, if there is a fix here, then maybe it is to more intuitively report which line of the file is in error. Mantid is indexing at 0, right? Whereas humans tend to index from 1!

Heh, this is an example of where the new changes to the error reporter would have been useful (so we could go back to the User and ask what they did)!

closed 2024-06-20 by github-actions[bot]