mantidproject/mantid #36573: Preview tab: retain ROI selections on loading run

rbauststfc, 2023-12-19:
**Is your feature request related to a problem? Please describe.**

On the Preview tab of the ISIS Reflectometry GUI you can select regions from both the instrument view and slice viewer plots. If you then load a new run these regions are cleared.

**Describe the solution you'd like**

Our scientists would prefer for any selected regions to remain when a new run is loaded. This wasn't requested, but we should presumably provide an easy way of clearing all regions for the case where the scientists do want to start from the beginning.

rbauststfc, 2025-02-10:
Getting this to work for the instrument view plot has the following complications:
 
1) We would need to identify from the experiment settings ROI detector string whether to add multiple region selectors or just one large one. We would need to be able to do this in a performant way that ideally didn't rely on a specific setup in the IDF (i.e. requiring use of a `RectangularDetector`) or a particular detector ID numbering sequence. We could save the string using a different pattern to allow us to distinguish separate regions, but this would require various changes, from how we retrieve the list of selected detectors from the instrument view (i.e. being able to get detector IDs grouped per mask shape rather than a single combined list from all the separate shapes) to how we parse it in the reduction.

1) We would need to be able to convert a detector ID to its corresponding pixel on the instrument view plot. There are some methods provided for this (see `AlFInstrumentWidget` example, which uses method `ProjectionSurface::drawShape2D`) but for various technical reasons we can't make use of these in the preview tab implementation. I think that investigating and implementing a resolution would likely be a fair amount of work.

1) If there is a switch to a per bank reduction in future then this will require a review of the preview tab to see how scientists will work with it when there is not longer a bank summing step. As a result changes that are relatively complex in nature are probably not advisable at this stage.

Some of the required changes on the instrument view side might be better covered under the instrument view epic.

I have confirmed with Max that if we can only add the selectors on the slice viewer plot in the short-term then that is still beneficial. A solution for the instrument view plot could come later.

MialLewis, 2025-04-03:
issue closed as completed, outstanding worked moved to issue @39147

closed 2025-04-03 by MialLewis