mantidproject/mantid #36589: Rename indirect interface files

robertapplin, 2023-12-22:
**Description:**
We want to rename some of the interfaces, and the classes/files related to these classes. This issue is an epic to keep track of this  rename of interface files.

- [ ] #36363  **Blocked**
- [ ] The `IndirectInterface` file should be renamed to `InelasticInterface`
- [ ] The files and classes in the `Inelastic/Analysis` folder should have `Indirect` removed from their name.
- [ ] The files and classes in the `Inelastic/Manipulation` folder should have `InelasticDataManipulation` removed from their name.
- [ ] The files and classes in the `Indirect/*` folders should have `Indirect` removed from their name.

closed 2024-05-09 by SilkeSchomann

sf1919, 2024-05-09:
@robertapplin @adriazalvarez should this have been closed?

robertapplin, 2024-05-09:
> @robertapplin @adriazalvarez should this have been closed?

Yes this should be complete now, thanks

sf1919, 2024-05-09:
I will change the milestone then

PR #37301 (2024-05-09): Remove "Indirect" from class names in Indirect/Inelastic
### Description of work
This PR renames the classes in Indirect/Inelastic scientific interface folder so that they no longer contain `Indirect` in their name. This is superflous info because the classes are already in the Indirect folder. This improves readability of these files.

The PR also renames IndirectTab and IndirectInterface to InelasticTab and InelasticInterface. This provides a better description.

Fixes #36589

### To test:
Check all these interfaces are available from the interface menu:
1. Indirect -> Data Reduction, Simula [... cut]