mantidproject/mantid #36690: Warning message when finishing running a script due to ADS.clear()

zjmorgan, 2024-01-23:
**Describe the bug**

If you create a workspace, mantid with throw a warning message at the end when exiting a script. This causes error messages to be reported in our autoreduction.

It produces this warning : `Running ADS.clear() also removes all hidden workspaces.
Mantid interfaces might still need some of these, for instance, MSlice.` 

**To Reproduce**

Run this test:
`mantidpythonnightly --classic test.py`

```python
from mantid.simpleapi import *
w = CreateSingleValuedWorkspace()
DeleteWorkspace(w)
```

Try it without creating a workspace and see no error
```python
from mantid.simpleapi import *
```

**Expected behavior**

Ideally, closing Mantid won't throw an error message.

**Screenshots**

[image]

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version nightly

**Additional context**

SilkeSchomann, 2024-01-24:
This was added to avoid problems described here: https://github.com/mantidproject/mslice/issues/839

zjmorgan, 2024-01-24:
Thanks. Unfortunately, the unintended consequence is a warning message is thrown after running any mantid script if any workspaces are created.

peterfpeterson, 2024-01-24:
If the [message appeared on stdout rather than stderr](https://github.com/mantidproject/mantid/pull/36548/files#diff-9bc82810fe60ea30c49fee293636f2576853f8374e58220ea808411a916df0c2R120-R121), I think the message wouldn't be perceived as an error by autoreduction.

A better solution is to only print the message if `ADS.clear()` was actually called at some point, but I imagine that would be much harder.

peterfpeterson, 2024-01-25:
I would also like to see that if the ADS is already empty, this message does not get printed at all. Another idea is to modify the [ADS atexit registration](https://github.com/mantidproject/mantid/blob/main/Framework/PythonInterface/mantid/api/src/Exports/AnalysisDataService.cpp#L41-L44) to pass along a variable that skips the warning too. Python shutting down suggests people don't care about things being deleted from the ADS.

It is a bit disconcerting to see this message appear every time mantid is closed and it would be good to reduce the number of times it is shown to the user.

closed 2024-01-31 by thomashampson