mantidproject/mantid #36692: Manual Testing Inelastic interfaces

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Follow the directions at:

* https://developer.mantidproject.org/Testing/Inelastic/DataAnalysisTests.html 
* https://developer.mantidproject.org/Testing/Inelastic/DataManipulationTests.html

MohamedAlmaki, 2024-01-24:
In the direct data Elwin section, the name of the workspaces is wrong it will produce `_red-276987691_red_elwin_eq2` instead of `MAR27691-27698_graphite002_red_elwin_eq2`

MohamedAlmaki, 2024-01-24:
Data analysis tab is now is in the inelastic section (documentation update)

MohamedAlmaki, 2024-01-24:
In the output section of the IQT tab, the units of the plots are always inverse angstrom even for the lifetimes

adriazalvarez, 2024-01-24:
Opened #36721 as a non regression. 
#36725 should be corrected before release. 

> In the output section of the IQT tab, the units of the plots are always inverse angstrom even for the lifetimes

Not sure how much this is an issue.

closed 2024-01-24 by adriazalvarez