mantidproject/mantid #36694: Manual Testing Workbench Group 1- 4

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

sf1919, 2024-01-24:
The information about this issue is incorrect. We no longer have groups 1-4 and the documentation that is linked also talked about grouped tests and the documentation not being up to date after v6.2

sf1919, 2024-01-24:
Using a conda install on osx the Help would not open. Just got an error message

sf1919, 2024-01-24:
Load MUSR00015189 and open a Contour plot on the whole group. Get the following error (Windows 10 only)

```Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\cbook\__init__.py", line 314, in process
    func(*args, **kwargs)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\mantid\plots\resampling_image\samplingimage.py", line 93, in _resize
    xbins, ybins = self._calculate_bins_from_extent()
  File "C:\MantidNightlyInstall\bin\lib\site-packages\mantid\plots\resampling_image\samplingimage.py", line 98, in _calculate_bins_from_extent
    bbox = self.get_window_extent().transformed(self.axes.get_figure().dpi_scale_trans.inverted())
AttributeError: 'NoneType' object has no attribute 'get_figure'
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\cbook\__init__.py", line 314, in process
    func(*args, **kwargs)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\mantid\plots\resampling_image\samplingimage.py", line 93, in _resize
    xbins, ybins = self._calculate_bins_from_extent()
  File "C:\MantidNightlyInstall\bin\lib\site-packages\mantid\plots\resampling_image\samplingimage.py", line 98, in _calculate_bins_from_extent
    bbox = self.get_window_extent().transformed(self.axes.get_figure().dpi_scale_trans.inverted())
AttributeError: 'NoneType' object has no attribute 'get_figure'```

sf1919, 2024-01-24:
We are seeing this one again #36096

thomashampson, 2024-01-24:
instrument view window title doesn't update when workspace name is changed

sf1919, 2024-01-24:
The script for setting scales, axes titles, plot titles etc on this page https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/plotting/01_basic_plot_scripting.html#basic-plot-scripting

For all OS when you run this script you get a line at the top but when you press home you get the plot you probably wanted! We wonder if the script needs updating rather than it being an issue with Mantid

sf1919, 2024-01-24:
The overplot on the same window script from the page above hangs on all OS. Maybe the script needs changing?

```
RawData = Load("MAR11015")

# Assign original plot to a window called graph_spce
graph_spec = plotSpectrum(RawData, 0)

# Overplot on that window, without clearing it
plotSpectrum(RawData, 1, window=graph_spec, clearWindow=False)
```

thomashampson, 2024-01-24:
> The overplot on the same window script from the page above hangs on all OS. Maybe the script needs changing?
>
> ```
> RawData = Load("MAR11015")
>
> # Assign original plot to a window called graph_spce [...]

After I restarted workbench, this worked. Then when I closed the plot and tried again it hangs.

thomashampson, 2024-01-24:
For me on IDAaaS my settings aren't persisted when I close and re-open workbench.

thomashampson, 2024-01-24:
following instruction here:
https://docs.mantidproject.org/nightly/plotting/index.html#custom-colormap-mantidworkbench
reveals a deprecation warning:
`The register_cmap function was deprecated in Matplotlib 3.7 and will be removed two minor releases later. Use ``matplotlib.colormaps.register(name)`` instead.`

sf1919, 2024-01-24:
ILL were unable to install the Script Repository on OSX - https://docs.mantidproject.org/nightly/workbench/scriptrepository.html#workbenchscriptrepository . Would @cailafinn be able to test it on an ISIS osx machine to make sure it's not facility dependent?

sf1919, 2024-01-24:
Problems with running Script B from this page - https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/solutions/03_pim_sol.html#pim-sol

On Windows and Linux we're finding it will freeze once it gets to the plotting part of the script.

thomashampson, 2024-01-24:
> Problems with running Script B from this page - https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/solutions/03_pim_sol.html#pim-sol
> 
> On Windows and Linux we're finding it will freeze once it gets to the plotting part of the script.

This looks to be the same as the other `plotSpectrum` related problem.  It seems to be intermittent.

thomashampson, 2024-01-24:
In the fit browser, once a parameter is fixed, it is no longer possible to remove the tie. After testing, I have determined that this was introduced in 6.7.

perenon, 2024-01-24:
> In the fit browser, once a parameter is fixed, it is no longer possible to remove the tie. After testing, I have determined that this was introduced in 6.7.

If you save the fit model (Setup => Manage Setup => Save Setup) and reload it (Setup => Custom Setup), the "Fix" becomes a "Tie", and then this constraint can be removed.

cailafinn, 2024-01-24:
> ILL were unable to install the Script Repository on OSX - https://docs.mantidproject.org/nightly/workbench/scriptrepository.html#workbenchscriptrepository . Would @cailafinn be able to test it on an ISIS osx machine to make sure it's not facility dependent?

Worked fine for me. macOS 14.1.1.

perenon, 2024-01-25:
> > ILL were unable to install the Script Repository on OSX - https://docs.mantidproject.org/nightly/workbench/scriptrepository.html#workbenchscriptrepository . Would @cailafinn be able to test it on an ISIS osx machine to make sure it's not facility dependent?
> 
> Worked fine for me. macOS 14.1.1.

Still failing on our side. Our macOS is 12.6. That is not a ILL-specific issue since it works on Ubuntu

cailafinn, 2024-01-25:
> Still failing on our side. Our macOS is 12.6. That is not a ILL-specific issue since it works on Ubuntu

Did it provide an error message when it failed? Would help nail down if there's anything else I need to do to reproduce.

perenon, 2024-01-25:
The error message is (please note the typo :-) ): ```ScriptRepository installation: Failed to download antidproject.org/scriptrepository/repository.json from <a href="https://download.mantidproject.org/scriptrepository/repository.json">https://download.mantidproject.org/scriptrepository/repository.json</a>.```

However, we have access to this URL through a browser.

closed 2024-01-26 by thomashampson