mantidproject/mantid #36695: Manual Testing Muon

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

See instructions at: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html

MohamedAlmaki, 2024-01-25:
In [tf symmetry test](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_Analysis_MUSR.html#transverse-field-asymmetry-test) we get a gil error when you change the fitting type to tf symmetry, remove the background and click fit

Stacktrace:
ConvertToHistogram-[Notice] ConvertToHistogram successful, Duration 0.00 seconds
ConvertToHistogram-[Debug] Unlocking MUSR62260; Group; fwd; Asymmetry; MA; Fitted; GausOsc,TFAsymmetry; Workspace; Simultaneous
Python-[Debug] using plotfunctions
Python-[Debug] using plotfunctions
Python-[Debug] using plotfunctions
Python-[Debug] using plotfunctions
Fatal Python error: PyThreadState_Get: the function must be called with the GIL held, but the GIL is released (the current Python thread state is NULL)
Python runtime state: initialized

Current thread 0x00007fa57c3ff640 (most recent call first):
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/basic_fitting/basic_fitting_model.py", line 723 in _create_fit_algorithm
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/general_fitting/general_fitting_model.py", line 340 in _do_simultaneous_fit_and_return_workspace_parameters_and_fit_function
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/general_fitting/general_fitting_model.py", line 331 in _do_simultaneous_fit
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/general_fitting/general_fitting_model.py", line 318 in perform_fit
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/tf_asymmetry_fitting/tf_asymmetry_fitting_model.py", line 543 in perform_fit
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqtinterfaces/Muon/GUI/Common/thread_model_wrapper.py", line 36 in execute
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/utils/asynchronous.py", line 63 in run
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 973 in _bootstrap

Thread 0x00007fa582d7f640 (most recent call first):
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 324 in wait
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 607 in wait
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 1376 in run
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 973 in _bootstrap

Thread 0x00007fa5a05ff640 (most recent call first):
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 324 in wait
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 607 in wait
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/mantidqt/utils/asynchronous.py", line 233 in inner_wrap
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 1378 in run
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 973 in _bootstrap

Thread 0x00007fa5a5fd1640 (most recent call first):
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/selectors.py", line 469 in select
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/asyncio/base_events.py", line 1871 in _run_once
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/asyncio/base_events.py", line 603 in run_forever
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/tornado/platform/asyncio.py", line 195 in start
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/ipykernel/iostream.py", line 92 in _thread_main
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 953 in run
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 973 in _bootstrap

Thread 0x00007fa5aa669640 (most recent call first):
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 320 in wait
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 607 in wait
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/IPython/core/history.py", line 894 in run
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/IPython/core/history.py", line 60 in only_when_enabled
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/site-packages/decorator.py", line 232 in fun
  File "/home/mohammed/Desktop/mantidworkbenchnightly/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
[... cut ...] t5 (total: 185)
Fontconfig error: Cannot load default config file: No such file: (null)
FrameworkManager-[Notice] Welcome to Mantid 6.8.20240123.1153
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
DownloadInstrument-[Notice] All instrument definitions up to date
CheckMantidVersion-[Notice] CheckMantidVersion started
CheckMantidVersion-[Notice] CheckMantidVersion successful, Duration 0.23 seconds
^A^A

jhaigh0, 2024-01-25:
In EMU [superconduction copper test](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_EMU.html#superconducting-copper-test), it instructs to go to the Model fitting tab. This no longer seems to exist in the Muon analysis interface.

jhaigh0, 2024-01-25:
In [FDA MaxEnt test](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_FDA.html#maxent-test), not seeing 5 plots as described.

jhaigh0, 2024-01-25:
For the [ARGUS tests](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_ARGUS.html) we were unable to complete as Mantid couldn't find the data (both linux and windows).

MohamedAlmaki, 2024-01-25:
All issues are raised.
Duration:
MUSR: 30m
PSI: 13m
HIFI: 12m
EMU: 13m
All: 1hr 51min

closed 2024-01-25 by MohamedAlmaki