mantidproject/mantid #36697: Manual Testing FilterEvents

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://docs.mantidproject.org/interfaces/utility/Filter%20Events.html

thomashampson, 2024-01-25:
in the "Maximum Value" box keep typing integers until it gets big enough to throw an exception.
or put a "-" in either that or the "minimum value" box

thomashampson, 2024-01-25:
No scripted testing, so we just tried to break it. We spent about 15 minutes.

thomashampson, 2024-01-25:
The issues we found were already reported last time.

closed 2024-01-25 by thomashampson