mantidproject/mantid #36698: Manual Testing Engineering Diffraction

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html

rbauststfc, 2024-01-24:
We took about 30 mins to get to the end of Test 4. Testing will continue from Test 5 tomorrow.

GuiMacielPereira, 2024-01-25:
- When doing Test 6, point 1, Loading a run does not automatically generate the background workspace with suffix `_bg`.

GuiMacielPereira, 2024-01-25:
- When doing Test 7, point 6, performing a fit through `Fit > Fit` closed the End Diff tab. When the tab was opened again, the fit had been successful and no data was deleted.

GuiMacielPereira, 2024-01-25:
- When doing Test 8, points 8-10, the order of the sequential fits did not follow the order of the log workspace `run_info_fitting`. Additionally, changing the primary log to blank also did not follow the order of the runs in the `Run Selection` table. In step 10, the order of the fits is indeed reversed from the order of the fits in step 8, but the table in `run_info_fitting` shows the same order.

GuiMacielPereira, 2024-01-25:
- When doing Test 10, point 8, with the range 20000-45000, Mantid crashed due to a missing file error.

GuiMacielPereira, 2024-01-25:
- In Test 10, point 10, `Refine in GSAS II` did not produce any output even after changing the time-out time to 120 seconds.

GuiMacielPereira, 2024-01-25:
- Completion time of the entire manual tests: 2h - 2:30h

jclarkeSTFC, 2024-01-26:
All issues opened, #36744, #36745, #36746, #36747, #36748

closed 2024-01-26 by jclarkeSTFC