mantidproject/mantid #36699: Manual Testing Sample Transmission calculator

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
See https://docs.mantidproject.org/interfaces/general/Sample%20Transmission%20Calculator.html

rbauststfc, 2024-01-24:
Some manual test instructions were added for this following feedback in the last period of manual testing, so we should follow: https://developer.mantidproject.org/Testing/General/SampleTransmissionCalculatorTestGuide.html.

I don't know if this link needs to be added to whatever generates these manual testing issues?

jclarkeSTFC, 2024-01-24:
> Some manual test instructions were added for this following feedback in the last period of manual testing, so we should follow: [...]

I'll update the link in the spreadsheet that generates the issues

jclarkeSTFC, 2024-01-24:
No other problems found.

closed 2024-01-24 by jclarkeSTFC

rbauststfc, 2024-01-24:
Testing took 25 mins