mantidproject/mantid #36700: Manual Testing ISIS Reflectometry

jclarkeSTFC, 2024-01-24:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

rbauststfc, 2024-01-24:
Polarisation Corrections instructions need to be clearer about which combo boxes are being referred to for selecting options (i.e. when we're talking about the Polarisation Corrections or Polarisation Efficiencies combo boxes).

warunawickramasingha, 2024-01-24:
When following below instructions, we did not notice the message with `ReflectometryReductionOneAuto` or the disabling/re-enabling of the tab

[Preview tab](https://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html#id16)

> Click the drop-down on the rectangle select button and select Transmission. Draw a transmission region onto the slice viewer plot. Then, in the same way, add one or more Background regions. The reduction should be re-run each time a region is added:
> 
> You should see the tab quickly disable and re-enable.
> 
> Another run of ReflectometryReductionOneAuto will be logged in the Messages bar.
> 
> The 1D plot should update (although this is usually only noticeable when changes are made to the Signal region).

rbauststfc, 2024-01-24:
The instructions need to be updated so that you're looking for another run of `ReflectometryISISLoadAndProcess` rather than `ReflectometryReductionOneAuto`.

rbauststfc, 2024-01-24:
Testing took 45 mins - 1hr

rbauststfc, 2024-01-24:
Known issue that the preview tab can crash as a consequence of interaction between slice viewer and preview tab region selection controls. Covered by existing issue #35347 and is not a regression. Requires more investigation to identify steps to replicate and fix - LSS will bump it up the priority list.

rbauststfc, 2024-01-24:
Closing as all follow-up issues have been created.

PR #37601 (2024-06-25): Finalise v6.10 release notes
### Description of work
A copy of #37600 but based on `release-next` branch. See that PR for further details.

#### Purpose of work
To fix #36700 for release

*There is no associated issue.*

### To test:
Check correct files have been deleted and spelling/grammar changes are correct.

All tests should pass

*This does not require release notes* because **it is cleaning up the release notes**

---

### Reviewer

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments will be used  [... cut]
closed 2024-01-24 by rbauststfc